ros2/rosbag2 #354: Undefined behaviors when exceeding the maximum recording capability

weihunko, 2020-04-06:
## Description
Actually I'm not sure if this should be a feature request or a but report, but we have seen some undefined behaviors (segfault, or not subscribe to all the topics) when trying to test the recording limitation of `rosbag2`.

### Expected Behavior
Gracefully exit if the bandwidth or number of topics exceeds the maximum loading of `rosbag2`.

### Actual Behavior
Segfault or only a subset of topics is discovered.

## To Reproduce
1. Prepare ROS 2 eloquent 
   1. https://index.ros.org/doc/ros2/Installation/Eloquent/Linux-Development-Setup/
   1. https://github.com/ros2/ros2/tree/release-eloquent-20200124
1. Clone and build [ros2_bandwidth_tester](https://github.com/weihunko/ros2_bandwidth_tester)
   1. `mkdir -p ~/ros2_bandwidth_tester_ws/src`
   1. `cd ~/ros2_bandwidth_tester_ws/src && git clone https://github.com/weihunko/ros2_bandwidth_tester`
   1. `cd ~/ros2_bandwidth_tester_ws`
   1. `colcon build`
1. In terminal 1: `ros2 bag record -a`
1. In terminal 2: `ros2 run ros2_bandwidth_tester bandwidth_tester --number 50 --size 200 --rate 100`
   1. This command will start 50 publishers, publishing strings with 200kb at rate 100Hz
1. At this point `ros2 bag record` will segfault
1. Tried the bag recording with different numbers of publishers, rates, and size of messages, sometimes the ros2 bag can still record but only discover a subset of all the existed publishers.

## System (please complete the following information)
 - OS: Ubuntu 18.04
 - ROS 2 Distro: Eloquent
 - Version: release-eloquent-20200124

## Additional context
N/A

dejanpan, 2020-04-09:
@Karsten1987 @emersonknapp LMK if this can be put on your board and reviewed in the Tooling WG meeting on April 10.

Karsten1987, 2020-04-09:
@dejanpan it will be part of the issue triaging. So yes, we'll discuss this during the meeting.
Or is there anything more to it you'd like to discuss as a separate agenda item?

dejanpan, 2020-04-10:
@Karsten1987 nope, issue triaging is enough for it to get on the radar.

emersonknapp, 2021-05-14:
There is still going to be some degradation at the limit of performacne - though we can make this better-defined, what happens in that case. Did the performance improvements since Foxy resolve this to a satisfactory state, or are there specific features we can implement to close this out?

MichaelOrlov, 2025-05-07:
- Closing this issue as a stale and not reproducible anymore.

closed 2025-05-07 by MichaelOrlov