ros2/rosbag2 #378: Export generic_subscription.hpp/cpp and generic_publisher.hpp/cpp for external use

TomyTomm, 2020-04-15:
## Description

I am working on creating a ROS cloud extension which will allow the customer to stream video and raw data to an external peer, and I would like it to allow for any ROS message types which the user may input. These files are currently located in the rosbag2_transport package.

Karsten1987, 2020-04-15:
I had this discussion with @emersonknapp in an earlier PR already. I personally think that this generic subscription is actually better suited within rclcpp than rosbag2. I don't think one has to be depend on rosbag2_transport for using this functionality. 

I am okay with making them public for Foxy though if that helps. @emersonknapp any thoughts? We can also make a push for rclcpp directly to see if that still lands in Foxy?

emersonknapp, 2020-04-15:
I think we're probably out of luck for including in rclcpp. The api freeze extension only applied to pre-planned work that slipped. I'd say or Foxy we expose from rosbag2 somewhere, and try to include it more centrally in G-turtle.

emersonknapp, 2020-04-15:
We could spin out a separate "generic transport" package within the rosbag2 repo, to make the distinction clear and the dependency small. The port into rclcpp later would be easy then

emersonknapp, 2020-04-15:
But, yeah, it may not be worth the extra complexity/effort right now.

Martin-Idel, 2020-04-15:
Just FYI: Here is the ticket in rclcpp which asks for that feature https://github.com/ros2/rclcpp/issues/544 - although the description is a bit too focused on the implementation.

The original plan was to make the split in rosbag such that one could just move the files and change some namespacing and such, but for various reasons we couldn't create it directly in rclcpp.

emersonknapp, 2020-04-24:
https://github.com/ros2/rclcpp/pull/1077
https://github.com/ros2/rclcpp/pull/1076

PRs in rclcpp to add this functionality there. This ticket wouldn't be closed until we moved the dependency over and removed the classes from rosbag2

emersonknapp, 2021-01-28:
https://github.com/ros2/rclcpp/pull/1452 is replacing the above PRs

Karsten1987, 2021-04-03:
closing with https://github.com/ros2/rclcpp/pull/1452

closed 2021-04-03 by Karsten1987