ros2/rosbag2 #380: Windows test_rosbag2_record_end_to_end test failure

emersonknapp, 2020-04-16:
## Description

The following tests are failing persistently on Windows CI builds

* record_end_to_end_with_splitting_bagsize_split_is_at_least_specified_size
* record_end_to_end_with_splitting_splits_bagfile

## System
 - OS: Ubuntu Focal
 - ROS 2 Distro: Foxy
 - Version: master, reporting as of 9909f8c3fd307d4

wjwwood, 2020-05-05:
> The following tests are failing persistently on Windows CI builds

> OS: Ubuntu Focal

Is this wrong? Btw, this test seems to timeout on Linux too:

https://ci.ros2.org/view/nightly/job/nightly_linux-aarch64_debug/1196/testReport/(root)/projectroot/test_rosbag2_play_end_to_end/

wjwwood, 2020-05-05:
Sorry, what I linked to is the `play` version, whereas this one is reporting the `record` one I think.

emersonknapp, 2020-05-05:
Yeah - that's right. It seems that `play_end_to_end` is now timing out on all platforms - not sure when exactly that started.

mm318, 2020-05-14:
Looks like https://github.com/ros2/rosbag2/pull/407 has possibly fixed this issue.

[Build 1618](https://ci.ros2.org/view/nightly/job/nightly_win_deb/1618/testReport/rosbag2_tests/RecordFixture/) was the last failing build. This issue has been fixed in [build 1620](https://ci.ros2.org/view/nightly/job/nightly_win_deb/1620/testReport/rosbag2_tests/RecordFixture/).

closed 2020-06-05 by mm318