ros2/rosbag2 #381: Periodic test failure on test_rosbag2_play_end_to_end

emersonknapp, 2020-04-16:
## Description

The regularly scheduled CI builds are seeing periodic failure of the test 

See the following overnight builds for example:
* https://github.com/ros2/rosbag2/runs/592733016?check_suite_focus=true
* https://github.com/ros2/rosbag2/runs/592032435?check_suite_focus=true

It does not fail every time, but has failed in approximately half of today's hourly runs.

The test times out when it fails, this suggests it is waiting for messages that are never delivered, or were delivered before the Subscriptions came online.

## System (please complete the following information)
 - OS: Ubuntu Bionic
 - ROS 2 Distro: Foxy
 - Version: master, reporting at cb44cdd1879508f5151

emersonknapp, 2020-05-01:
From the last day or so these are the test results, it's not always the same tests failing. The list of unstable tests seems to include:

* test_play_topic_remap__rmw_fastrtps_cpp (Timeout)
* test_play_topic_remap__rmw_fastrtps_dynamic_cpp (Timeout)

* test_record__rmw_fastrtps_cpp (Failed)
* test_record__rmw_fastrtps_dynamic_cpp (Failed)
* test_record__rmw_cyclonedds_cpp (Failed)
* test_play_topic_remap__rmw_fastrtps_dynamic_cpp (Failed)

* test_rosbag2_play_end_to_end (Timeout)

The timeouts probably have something to do with discovery, or shutdown behavior not receiving the final message published by a publisher.

The failures, I'm guessing are segmentation faults of some kind. It seems common with fastrtps, and occurred once in cyclone in this list.

Here's my output from the script I used to scrape, for context:

```
TEST RUN 93323352 - 2020-05-01T19:03:53Z
b'2020-05-01T19:38:31.9162579Z The following tests FAILED:'
b'2020-05-01T19:38:31.9162907Z \t 22 - test_play_topic_remap__rmw_fastrtps_cpp (Timeout)'
b'2020-05-01T19:38:31.9163376Z \t 30 - test_record__rmw_fastrtps_dynamic_cpp (Failed)'
b'2020-05-01T19:38:31.9164216Z \t 33 - test_play_topic_remap__rmw_fastrtps_dynamic_cpp (Timeout)'
b'2020-05-01T19:38:31.9164349Z Errors while running CTest'

b'2020-05-01T19:40:03.0866840Z The following tests FAILED:'
b'2020-05-01T19:40:03.0867133Z \t  2 - test_rosbag2_play_end_to_end (Timeout)'
b'2020-05-01T19:40:03.0867235Z Errors while running CTest'

TEST RUN 93280328 - 2020-05-01T18:04:10Z
b'2020-05-01T18:35:20.8619814Z The following tests FAILED:'
b'2020-05-01T18:35:20.8620177Z \t 33 - test_play_topic_remap__rmw_fastrtps_dynamic_cpp (Timeout)'
b'2020-05-01T18:35:20.8620287Z Errors while running CTest'

b'2020-05-01T18:36:51.5043900Z The following tests FAILED:'
b'2020-05-01T18:36:51.5044307Z \t  2 - test_rosbag2_play_end_to_end (Timeout)'
b'2020-05-01T18:36:51.5044399Z Errors while running CTest'

TEST RUN 93238438 - 2020-05-01T17:03:56Z
b'2020-05-01T17:32:03.5810154Z The following tests FAILED:'
b'2020-05-01T17:32:03.5810416Z \t 19 - test_record__rmw_fastrtps_cpp (Failed)'
b'2020-05-01T17:32:03.5810677Z \t 22 - test_play_topic_remap__rmw_fastrtps_cpp (Timeout)'
b'2020-05-01T17:32:03.5810956Z \t 33 - test_play_topic_remap__rmw_fastrtps_dynamic_cpp (Timeout)'
b'2020-05-01T17:32:03.5811165Z Errors while running CTest'

b'2020-05-01T17:33:33.1755227Z The following tests FAILED:'
b'2020-05-01T17:33:33.1755510Z \t  2 - test_rosbag2_play_end_to_end (Timeout)'
b'2020-05-01T17:33:33.1755601Z Errors while running CTest'

TEST RUN 93196491 - 2020-05-01T16:04:10Z
b'2020-05-01T16:33:03.8155439Z The following tests FAILED:'
b'2020-05-01T16:33:03.8155720Z \t 22 - test_play_topic_remap__rmw_fastrtps_cpp (Timeout)'
b'2020-05-01T16:33:03.8156111Z \t 33 - test_play_topic_remap__rmw_fastrtps_dynamic_cpp (Timeout)'
b'2020-05-01T16:33:03.8156235Z Errors while running CTest'

b'2020-05-01T16:34:32.4372036Z The following tests FAILED:'
b'2020-05-01T16:34:32.4372243Z \t  2 - test_rosbag2_play_end_to_end (Timeout)'
b'2020-05-01T16:34:32.4372335Z Errors while running CTest'

TEST RUN 93154036 - 2020-05-01T15:03:52Z
b'2020-05-01T15:37:13.5761261Z The following tests FAILED:'
b'2020-05-01T15:37:13.5761525Z \t 19 - test_record__rmw_fastrtps_cpp (Failed)'
b'2020-05-01T15:37:13.5761775Z \t 22 - test_play_topic_remap__rmw_fastrtps_cpp (Timeout)'
b'2020-05-01T15:37:13.5762050Z \t 33 - test_play_topic_remap__rmw_fastrtps_dynamic_cpp (Timeout)'
b'2020-05-01T15:37:13.5762165Z Errors while running CTest'

b'2020-05-01T15:38:43.8400797Z The following tests FAILED:'
b'2020-05-01T15:38:43.8401031Z \t  2 - test_rosbag2_play_end_to_end (Timeout)'
b'2020-05-01T15:38:43.8401136Z Errors while running CTest'
[... 116 lines cut ...]
b'2020-05-01T02:39:04.3343220Z The following tests FAILED:'
b'2020-05-01T02:39:04.3343460Z \t  2 - test_rosbag2_play_end_to_end (Timeout)'
b'2020-05-01T02:39:04.3343568Z Errors while running CTest'

TEST RUN 92691030 - 2020-05-01T01:04:05Z
b'2020-05-01T01:34:46.3406834Z The following tests FAILED:'
b'2020-05-01T01:34:46.3407183Z \t 20 - test_play__rmw_fastrtps_cpp (Failed)'
b'2020-05-01T01:34:46.3407465Z \t 22 - test_play_topic_remap__rmw_fastrtps_cpp (Timeout)'
b'2020-05-01T01:34:46.3407723Z \t 33 - test_play_topic_remap__rmw_fastrtps_dynamic_cpp (Timeout)'
b'2020-05-01T01:34:46.3407830Z Errors while running CTest'

b'2020-05-01T01:36:16.3433446Z The following tests FAILED:'
b'2020-05-01T01:36:16.3433669Z \t  2 - test_rosbag2_play_end_to_end (Timeout)'
b'2020-05-01T01:36:16.3433769Z Errors while running CTest'

TEST RUN 92656135 - 2020-05-01T00:09:17Z
b'2020-05-01T00:38:39.7218986Z The following tests FAILED:'
b'2020-05-01T00:38:39.7219224Z \t  2 - test_rosbag2_play_end_to_end (Timeout)'
b'2020-05-01T00:38:39.7219330Z Errors while running CTest'

TEST RUN 92613255 - 2020-04-30T23:03:53Z
b'2020-04-30T23:33:43.3972138Z The following tests FAILED:'
b'2020-04-30T23:33:43.3972601Z \t 33 - test_play_topic_remap__rmw_fastrtps_dynamic_cpp (Timeout)'
b'2020-04-30T23:33:43.3980234Z Errors while running CTest'
```

mabelzhang, 2020-05-01:
Is this on Bionic? I saw similar fastrtps test failures in rosbag2 when I was on Bionic. They disappeared when I started compiling in Focal instead.

emersonknapp, 2020-05-01:
Yeah it's running on Bionic currently - which it should not. I still feel that the first most important change is to get these running on Focal. However it's interesting that Karsten can't repro on his Bionic machine. But, it's not a supported platform so it's not worth tooo much effort tracking it down if it's working on Focal. I have a feeling that a few of these issues will linger on Focal, so the faster we can get that running the better so we can track the remainder down.

This is a prerequisite to getting us there https://github.com/ros-tooling/setup-ros/issues/140

emersonknapp, 2020-05-18:
Still seeing Action CI fail consistently e.g. 
* https://github.com/ros2/rosbag2/pull/411/checks
* https://github.com/ros2/rosbag2/actions/runs/108446931 

so I'm reopening this to track getting us to green

mm318, 2020-05-21:
I have been able to reproduce these issues locally in a Ubuntu Focal container.

The segfaults (for example in `test_rosbag2_record_end_to_end`, `test_record__*`, `test_record_all_no_discovery__*`) occur during `rclcpp::Node` construction and `rclcpp::shutdown()`. Since these tests seem to involve quite a bit of asynchronous code execution, I suspect the issues might be related to https://github.com/ros2/rclcpp/issues/1042.

The timeouts happened in `test_rosbag2_play_end_to_end` and `test_play_topic_remap__rmw_fastrtps_cpp`, which may mean it is specific to when using `rmw_fastrtps_cpp`/`rmw_fastrtps_dynamic_cpp` as the RMW implementation.

emersonknapp, 2020-05-21:
I assume these timeouts are related to expecting a specific number of messages, and not receiving either the messages at the beginning, or one at the end - it's worth putting a print statement on messages received, to see if it's just that we're getting one fewer than expected. So, either a discovery or a shutdown edge case behavior - if the tests are testing something else (e.g. _not_ about specific numbers of messages, but about whether messages are received at all, etc), then what we should do is make those tests less fragile to these boundary behaviors, perhaps publish N*2 messages and only expect to receive N - or something along those lines.

mm318, 2020-05-21:
> I assume these timeouts are related to expecting a specific number of messages, and not receiving either the messages at the beginning, or one at the end

In both tests that failed due to timeout that I mentioned, there were indeed an expectation for a specific number of messages, but exactly 0 messages were received (I used the debugger instead of print statements). I think we're missing waiting for `get_publisher_count()` of the `SubscriptionManager` and the `get_subscription_count()` of the `Rosbag2Transport`/`ros2 bag play` to reach non-zero. It could also be that `get_publisher_count()`/`get_subscription_count()` will never be non-zero due to some discovery edge case behavior as you mentioned.

mm318, 2020-06-10:
I wrote a small case that reproduces the timeout (messages never actually received by subscription) problem at [unpublished_messages.zip](https://github.com/ros2/rosbag2/files/4756269/unpublished_messages.zip).

emersonknapp, 2020-06-10:
That's awesome! With this we can potentially report it to the rmw implementation - does it show this behavior only on Fast-RTPS?

emersonknapp, 2020-12-07:
Bump - watch this. The Action CI has been reenabled, see how it acts over the next 24 hours before closing this issue

emersonknapp, 2020-12-08:
only one of the last 24 hourly runs failed - and that test failure is also seen on the buildfarm. followup work to de-flake will be tracked by https://github.com/ros2/rosbag2/issues/408

PR #396 (2020-05-12): Fix #381 unstable play_end_to_end test
I think this should fix #381. The failing test was `play_filters_by_topic` in `rosbag2_tests`.

I took the same kind of fix from `rosbag2_transport` https://github.com/ros2/rosbag2/blob/master/rosbag2_transport/test/rosbag2_transport/test_play.cpp#L140 , so I think it should make the tests more stable.

I ran GitHub Actions on this branch in my fork (temporarily the default branch to run the scheduled Actions) https://github.com/mabelzhang/rosbag2/actions , and the ros-tooling/action-ros-ci passed most of the time. The occasional times it fails are at `am [... cut]
commit cf199290 (2020-05-12): Fix #381 unstable play_end_to_end test (#396)

* decrease subscription expected msgs

* change workflow to test this branch

* change expected number too

* linter

* change workflow to autotest feature branch

* revert workflow

* run workflow more frequently to test stability

* revert github workflow

* cleanup

* define constants

* disable test on windows

* disable play_filters_by_topic on windows
closed 2020-12-08 by emersonknapp