ros2/rosbag2 #393: jacob/rosbag2_py branch sqlite_storage.cpp:70:5: error: ‘relative_path_’ was not declared in this scope

WaruWang, 2020-04-28:
Tried both ROS crystal and dashing release, errors are same

rosbag_ws/src/rosbag2/rosbag2_storage_default_plugins/src/rosbag2_storage_default_plugins/sqlite/sqlite_storage.cpp:70:5: error: ‘relative_path_’ was not declared in this scope
sqlite_storage.cpp:155:44: error: no ‘uint64_t rosbag2_storage_plugins::SqliteStorage::get_bagfile_size() const’ member function declared in class ‘rosbag2_storage_plugins::SqliteStorage’
sqlite_storage.cpp:191:78: error: no ‘void rosbag2_storage_plugins::SqliteStorage::remove_topic(const rosbag2_storage::TopicMetadata&)’ member function declared in class ‘rosbag2_storage_plugins::SqliteStorage’
sqlite_storage.cpp:237:53: error: no ‘std::__cxx11::string rosbag2_storage_plugins::SqliteStorage::get_relative_file_path() const’ member function declared in class ‘rosbag2_storage_plugins::SqliteStorage’
/sqlite_storage.cpp:242:55: error: no ‘uint64_t rosbag2_storage_plugins::SqliteStorage::get_minimum_split_file_size() const’ member function declared in class ‘rosbag2_storage_plugins::SqliteStorage’

emersonknapp, 2020-04-29:
I'm sorry, I don't fully understand what you're showing here - ideally could you use the issue template to provide more structured info?

Minimal info needed - what did you do to get this error? Did you build `master` branch of `rosbag2` from source against an installed previous ROS 2 distro?

WaruWang, 2020-04-30:
I am looking for a solution which can generate csv format of rosbag2 like ros1.
I tried twice on U18.04, first time I installed ros2 dashing and 2nd time I installed ros2 crystal. And I could build rosbag2 master branch code with them. 
But when I tried to build jacob/rosbag2_py branch, I got errors above. 
So I wonder is jacob/rosbag2_py branch ready for use or should I do something special?

emersonknapp, 2020-04-30:
Oh - the only officially supported branches are
* master
* crystal
* dashing
* eloquent

PRs are welcome if you'd like to try and introduce the features from that branch in to master

jacobperron, 2020-06-18:
I pushed some prototype work to the branch [jacob/rosbag2_py](https://github.com/ros2/rosbag2/tree/jacob/rosbag2_py). It was not ready for general use. It has been extended, and currently being iterated on as part of https://github.com/ros2/rosbag2/pull/308.

closed 2020-06-18 by jacobperron