ros2/rosbag2 #402: make rosbag2 related packages relocatable

dirk-thomas, 2020-05-04:
Atm various `.cmake` files installed by `rosbag2_*` packages and its dependencies contain absolute paths to the install space which makes them non-relocatable. See:

* https://github.com/ros2/ros2/issues/915#issuecomment-623552072
* https://github.com/ros2/ros2/issues/904

dirk-thomas, 2020-05-04:
FYI https://github.com/ros2/console_bridge_vendor/pull/9#discussion_r419693928

Karsten1987, 2020-05-06:
With the referenced PRs, the exported cmake targets look good now for rosbag2_transport (the highest package in the hierarchy):

```
set_target_properties(rosbag2_transport::rosbag2_transport PROPERTIES
  INTERFACE_INCLUDE_DIRECTORIES "${_IMPORT_PREFIX}/include"
  INTERFACE_LINK_LIBRARIES "rcl::rcl;rclcpp::rclcpp;rcutils::rcutils;rmw::rmw;rosbag2_compression::rosbag2_compression;rosbag2_compression::rosbag2_compression_zstd;rosbag2_cpp::rosbag2_cpp;shared_queues_vendor::singleproducerconsumer;shared_queues_vendor::concurrentqueue;yaml-cpp"
)
```

Karsten1987, 2020-05-06:
Isolated packaging job:
[image]

Karsten1987, 2020-05-06:
new round of CI:
[image]

Karsten1987, 2020-05-06:
So I downloaded the package job linked above and looked at the content of exported cmake files of rosbag2_transport.

```
set_target_properties(rosbag2_transport::rosbag2_transport PROPERTIES
  INTERFACE_INCLUDE_DIRECTORIES "${_IMPORT_PREFIX}/include"
  INTERFACE_LINK_LIBRARIES "rcl::rcl;rclcpp::rclcpp;rcutils::rcutils;rmw::rmw;rosbag2_compression::rosbag2_compression;rosbag2_compression::rosbag2_compression_zstd;rosbag2_cpp::rosbag2_cpp;shared_queues_vendor::singleproducerconsumer;shared_queues_vendor::concurrentqueue;yaml-cpp"
)
```

So there's no absolute path in it which makes me believe it's in a decent state. I can't really use it locally on my machine as I don't have a Focal installation at hand. If somebody could verify that the packages are indeed now relocatable, that'd be great.

@dirk-thomas maybe you could have a look at this. I am deliberately not running a full set of CI yet because I want to rule out that a) this is actually needed and b) what kind of jobs I should run. I could imagine that it could make sense to run a debug/release job to make sure that the imported libraries are correctly set.

Opinions?

dirk-thomas, 2020-05-06:
> maybe you could have a look at this.

Please try it in a Focal VM or a Docker container instead.

> I am deliberately not running a full set of CI yet because I want to rule out that a) this is actually needed and b) what kind of jobs I should run.

At least a build on all platforms makes sense.

Karsten1987, 2020-05-06:
CI, building all, testing `rosbag2_transport`.
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

dirk-thomas, 2020-05-06:
You might want to test all packages touched in order to run their linters.

Karsten1987, 2020-05-06:
building all, testing `rosbag2_storage rosbag2_transport rosbag2_cpp rosbag2_test_common rosbag2_tests pluginlib tinyxml2_vendor yaml_cpp_vendor rviz_common`
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

Karsten1987, 2020-05-13:
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

Karsten1987, 2020-05-13:
I think this is good to go for a review.

dirk-thomas, 2020-05-13:
Please enumerate the PRs included in the above CI builds.

Karsten1987, 2020-05-13:
It includes all the ones which are linked here to this ticket.

https://github.com/ros2/rosbag2/pull/403
https://github.com/ros2/tinyxml2_vendor/pull/8
https://github.com/ros2/yaml_cpp_vendor/pull/9
https://github.com/ros/pluginlib/pull/190
https://github.com/ros2/rviz/pull/542

Karsten1987, 2020-05-14:
Another round of CI with the latest changes from #403 
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

Karsten1987, 2020-05-14:
My only explanation for the windows build failure is that the package wasn't rebased correctly - it passes locally in my VM:

another shot: 
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

Karsten1987, 2020-05-15:
After [RViz magic](https://github.com/ros2/rviz/pull/542/commits/b44d6606cd10ea217935e101bc2d06f3955cb8d7), another round of CI:

* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

Karsten1987, 2020-05-15:
RViz magic, part two, another round of CI:
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

Karsten1987, 2020-05-20:
repos file was outdated:
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

PR #403 (2020-05-20): export targets
Closes #402 
Requires https://github.com/ros/pluginlib/pull/190
closed 2020-05-20 by Karsten1987