ros2/rosbag2 #404: rosbag2_compression relies on relative header in another package

jacobperron, 2020-05-06:
## Description

rosbag2_compression includes relative headers from a different package:

https://github.com/ros2/rosbag2/blob/69c58939bc5a873c6a25b7e36af81f9410c48ce0/rosbag2_compression/test/rosbag2_compression/test_sequential_compression_reader.cpp#L26-L29

This causes a compilation error in packaging jobs:

```
17:29:00 /tmp/binarydeb/ros-foxy-rosbag2-compression-0.2.5/test/rosbag2_compression/test_sequential_compression_reader.cpp:26:10: fatal error: ../../rosbag2_cpp/test/rosbag2_cpp/mock_converter_factory.hpp: No such file or directory
17:29:00    26 | #include "../../rosbag2_cpp/test/rosbag2_cpp/mock_converter_factory.hpp"
17:29:00       |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

The bug is causing the Debian packaging job to fail: http://build.ros2.org/view/Fbin_uF64/job/Fbin_uF64__rosbag2_compression__ubuntu_focal_amd64__binary/5/

To remove these relative includes, we should either duplicate the needed mock objects in rosbag2_compression or install them to be shared by multiple packages.

Karsten1987, 2020-05-08:
clsoed via #405

closed 2020-05-08 by Karsten1987

jacobperron, 2020-05-08:
Thanks @zmichaels11 and @Karsten1987 !