ros2/rosbag2 #408: test_rosbag2_play_end_to_end flaky

tfoote, 2020-05-11:
## Description
The test regressed overnight. 

https://ci.ros2.org/job/nightly_linux_release/1539/testReport/(root)/projectroot/test_rosbag2_play_end_to_end/

```
-- run_test.py: invoking following command in '/home/jenkins-agent/workspace/nightly_linux_release/ws/build/rosbag2_tests':
 - /home/jenkins-agent/workspace/nightly_linux_release/ws/build/rosbag2_tests/test_rosbag2_play_end_to_end --gtest_output=xml:/home/jenkins-agent/workspace/nightly_linux_release/ws/build/rosbag2_tests/test_results/rosbag2_tests/test_rosbag2_play_end_to_end.gtest.xml
Running main() from gmock_main.cc
[==========] Running 4 tests from 1 test case.
[----------] Global test environment set-up.
[----------] 4 tests from PlayEndToEndTestFixture
[ RUN      ] PlayEndToEndTestFixture.play_end_to_end_test
[INFO] [1589203781.466639626] [rosbag2_storage]: Opened database 'cdr_test/cdr_test_0.db3' for READ_ONLY.
[       OK ] PlayEndToEndTestFixture.play_end_to_end_test (612 ms)
[ RUN      ] PlayEndToEndTestFixture.play_fails_gracefully_if_bag_does_not_exist
[       OK ] PlayEndToEndTestFixture.play_fails_gracefully_if_bag_does_not_exist (237 ms)
[ RUN      ] PlayEndToEndTestFixture.play_fails_gracefully_if_needed_coverter_plugin_does_not_exist
[       OK ] PlayEndToEndTestFixture.play_fails_gracefully_if_needed_coverter_plugin_does_not_exist (283 ms)
[ RUN      ] PlayEndToEndTestFixture.play_filters_by_topic
[INFO] [1589203782.518986655] [rosbag2_storage]: Opened database 'cdr_test/cdr_test_0.db3' for READ_ONLY.
[WARN] [1589203782.746890923] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[INFO] [1589203782.999761379] [rosbag2_storage]: Opened database 'cdr_test/cdr_test_0.db3' for READ_ONLY.
```

Looking at the history it has failed 4 out of 7 of the last buils. 

https://ci.ros2.org/job/nightly_linux_release/1539/testReport/(root)/projectroot/test_rosbag2_play_end_to_end/history/

If this cannot be resolved immediately I will plan to add an xfail flag to this to avoid busting CI and this ticket can stay open to track the failure.

Karsten1987, 2020-05-11:
This might be fixed by https://github.com/ros2/rosbag2/pull/396
/cc @mabelzhang

tfoote, 2020-05-14:
Follow up to #396 in #410

wjwwood, 2020-05-27:
Should this be closed now?

mabelzhang, 2020-05-27:
We just disabled the problematic test. We could see if it can be fixed once the pause feature is in for rosbag2. I don't know if we want to leave this open until then or not.

emersonknapp, 2020-07-17:
The test is currently disabled - the resolution for this ticket is to fix its stability and re-enable.

emersonknapp, 2020-12-08:
We still see an occasional timeout on this test - https://github.com/ros2/rosbag2/runs/1515318271

PR #1297 (2023-05-12): Address flakiness in rosbag2_play_end_to_end tests
- Closes #408 
Make test_rosbag2_play_end_to_end more deterministic
- Change QoS depth in test databases to correspond number of messages
- Change QoS durability to transient local in test DB and mcap file
- Explicitly specify QoS depth = 10 for subscribers
- Explicitly specify QoS reliability to reliable for subscribers
- Explicitly specify QoS durability to transient local for subscribers
- Update metadata in test DB and mcap files to the latest version(7)
- Start player in pause mode and wait on subscribers for matched
publishers from playe [... cut]
closed 2023-05-12 by MichaelOrlov