ros2/rosbag2 #412: Playing a bag recorded with file compression segfaults

audrow, 2020-05-15:
## Description
When I try to play a bag that has been recorded with the compression mode `zstd` and format `file`, I receive a segmentation fault when I try to play that bag.

### Expected Behavior
I expect to have the bag play.

### Actual Behavior
The bag seems to open and then a segmentation fault occurs.  Here is the printout.
```
[INFO] [1589568927.962724388] [rosbag2_storage]: Opened database 'compressed_bag/compressed_bag_0.db3' for READ_ONLY.
Segmentation fault (core dumped)
```

## To Reproduce
0. In one terminal, create a publisher:
   ```
   ros2 topic pub /foo std_msgs/msg/String data:\ \'Hi\'\ 
   ```
1. In another terminal, record a bag:
   ```
   ros2 bag record --compression-format zstd --compression-mode file --output compressed_bag /foo
   ```
2. Hit `Ctrl+C` to stop recording the bag.

3. Try to play the bag:
   ```
   ros2 bag play compressed_bag
   ```

## System
 - OS: Ubuntu Focal
 - ROS 2 Distro: Foxy
 - Version: 0.2.7-1focal.20200514.041002

emersonknapp, 2020-05-18:
I am starting to look into this now. Thanks for reporting.

closed 2020-06-18 by zmichaels11

Karsten1987, 2020-06-18:
@zmichaels11 what changed in between that you closed this ticket? Same for #413

zmichaels11, 2020-06-18:
@Karsten1987 our team reviewed the work done and verified that the problem was solved.
@emersonknapp can you add a comment here explaining why?

emersonknapp, 2020-06-18:
This was resolved with https://github.com/ros2/rosbag2/pull/417 but our internal process had blocked actually closing the issue pending team review/demo. We're going to be modifying that process though. Sorry for any confusion.