ros2/rosbag2 #414: Compress by message

emersonknapp, 2020-05-19:
## Description

Compress individual rosbag messages rather than entire bagfiles.

## Related Issues

Part of https://github.com/ros2/rosbag2/issues/152

## Completion Criteria

Individual messages in a non-compressed bag file are compressed.

## Implementation Notes / Suggestions

** If you have ideas about how this feature might be accomplished, put them here. Note that this is just a suggestion to the implementer, so feel free to speculate. **

## Testing Notes / Suggestions

** All features in this project need tests. Please give some input on cases that will need to be tested - and how the testing might be implemented. **

pjreed, 2020-05-20:
This is something that would be very useful for me, and I'd like to work on it if nobody else has expressed interest yet.

emersonknapp, 2020-05-20:
Sounds great! I've assigned the task to you, let me know if the situation changes at all :)

pjreed, 2020-05-20:
Thanks!

I'm looking through the code now to get an idea of how it all works, and one thing I noticed is that there doesn't seem to be any mechanism in rosbag2_storage_plugins::SqliteStorage for versioning database schemas or migrating between them.  Is that also something we should worry about, or is the use case of wanting to upgrade an existing bag file to a newer version rare enough that we don't want to worry about it?

emersonknapp, 2020-05-20:
I'm not sure that we're handling database schema versioning at all - I think that so far there have not been any incompatible schema changes to the DB. @Karsten1987 knows more about this, though, and can probably correct that statement.

The metadata yaml file handles schema versioning by simple switch statements (which will likely not scale well if there are frequent updates to the metadata schema but for now it's a minimal amount of logic)

pjreed, 2020-05-28:
Good news!  This is now working.  I've been able to successfully record and play back bags with per-message zstd compression.

There are a few other things I've been thinking about that might be worth considering, though.  Right now, `MESSAGE` is a compression type that is set for the whole bag file in the metadata.  It would be nice if compression could be enabled or disabled on a per-message basis.  This would make it easier to enable #275, where rosbag could automatically determine whether to compress messages or not based on whether it's keeping up with its throughput.  Another possibility would be to enable compression on a per-topic basis; it could be useful to selectively enable or disable compression based a on a priori knowledge of how well a topic type can be compressed.

The best way I've thought of to do that, though, would require modifying the database schema to add a new column to the message table, but doing that properly would also mean needing to version the database schema, handle migrating to the new version if appending to an existing bag, or being able to read old bags that aren't using the new schema.  That might be a decent amount of work that could be done in a separate issue.

Any other thoughts on the matter?

emersonknapp, 2020-05-28:
> It would be nice if compression could be enabled or disabled on a per-message basis

Like you've said, I think that anything further would require a good deal more code complexity and should maybe have some design discussion around it.

I don't think we need a complete solution now, as long as what we put in is better than before  without limiting our options in the future, then I'm not too worried about it. The "whole-bag message compression" option could continue to exist while adding narrower options for topics/types/performance-cases, so my opinion is that your implementation will fit that philosophy and we should move forward with it.

pjreed, 2020-06-03:
Ok, I think this solution is ready to go for now.

PR #418 (2020-07-13): Add per-message ZSTD compression
Fixes #414.

This implements the per-messages compression and decompression
functions for the ZSTD compressor and also adds unit tests
for them.

Distro A, OPSEC #2893
closed 2020-07-13 by piraka9011