ros2/rosbag2 #425: DDS Node Name not unique

dawonn-haval, 2020-06-01:
## Description
The Node Name is always _ros2cli_rosbag2, but the DDS spec requires unique names, _I think_.  

I'm debugging other DDS issues and happened to notice this. I'm not sure if it's a real problem or not... Other ros2cli nodes will add their PID as a suffix to 'ensure' uniqueness. (This clearly wont scale to multiple hosts, but it's a start.)

### Expected Behavior
Unique Node Names.

### Actual Behavior
```
$ ros2 topic info -v /str1
Type: std_msgs/msg/String

Publisher count: 1

Node name: _ros2cli_554
Node namespace: /
Topic type: std_msgs/msg/String
Endpoint type: PUBLISHER
GID: 01.0f.1b.26.2a.02.00.00.01.00.00.00.00.00.05.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RMW_QOS_POLICY_RELIABILITY_RELIABLE
  Durability: RMW_QOS_POLICY_DURABILITY_TRANSIENT_LOCAL
  Lifespan: 2147483651294967295 nanoseconds
  Deadline: 2147483651294967295 nanoseconds
  Liveliness: RMW_QOS_POLICY_LIVELINESS_AUTOMATIC
  Liveliness lease duration: 2147483651294967295 nanoseconds

Subscription count: 1

Node name: _ros2cli_rosbag2
Node namespace: /
Topic type: std_msgs/msg/String
Endpoint type: SUBSCRIPTION
GID: 01.0f.bd.21.d9.50.00.00.01.00.00.00.00.00.12.04.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RMW_QOS_POLICY_RELIABILITY_RELIABLE
  Durability: RMW_QOS_POLICY_DURABILITY_TRANSIENT_LOCAL
  Lifespan: 2147483651294967295 nanoseconds
  Deadline: 2147483651294967295 nanoseconds
  Liveliness: RMW_QOS_POLICY_LIVELINESS_AUTOMATIC
  Liveliness lease duration: 2147483651294967295 nanoseconds
```

## To Reproduce
run multiple instances of ros bag. I have one that records 'vital' information for reports, and another that records sensor data for post-mortem debugging. 

## System (please complete the following information)
 - OS: Ubuntu 20.04
 - ROS 2 Distro: Foxy
 - Version: bfa63199421e53b8d229b9403ac7dd4ad4f155cb

## Additional context
Again, I'm not sure if this is a real bug. I just happened to notice it and wasn't sure of a better forum to ask. :)

emersonknapp, 2020-06-01:
I'm inclined to close this as "not an issue" - DDS doesn't require unique node names, and ROS 2 has no mechanism for it either (yet) - conversation around it in https://github.com/ros2/design/issues/187

emersonknapp, 2020-06-01:
However... suffixing the process ID might be a fine idea, so maybe we can keep this open as an enhancement. These node names for cli tools are all very "under the hood" anyways, so users don't really interact with them. Since DDS is anonymous pub-sub, the only place where this really comes into play is if we expose Services or Actions from the rosbag2 node, because the node name then becomes part of a topic names, which _does_ affect behavior.

emersonknapp, 2023-07-25:
Yes, there are a variety of exposed services, and they all are an issue with non-unique names. I'll note that since https://github.com/ros2/rosbag2/pull/1180, users can now specify `--node-name` to manually choose a unique name for their nodes (one could even use e.g. [`uuid`](https://docs.python.org/3/library/uuid.html) to generate them in python launchfiles). So at least there is a way to accomplish this.

However, the solution is (I think) so simple, that I'm going to put up a draft PR momentarily to address this.