ros2/rosbag2 #431: Negative values allowed in overrides file

dabonnie, 2020-06-04:
## Description
When defining an [overrides file](https://index.ros.org/doc/ros2/Tutorials/Ros2bag/Overriding-QoS-Policies-For-Recording-And-Playback/) negative values are allowed for the following fields:
- depth 
- deadline
 - lifespan
 - liveliness_lease_duration

when using `ros2 bag record` or `ros2 bag play` 

### Expected Behavior
The CLI throws an error / exception stating that negative values are not allowed for these fields.

### Actual Behavior
The CLI accepts the values and attempts to record / play.

## To Reproduce
** Steps to reproduce the behavior, e.g.
1. Define an overrides file with negative values, e.g., 

```
/test:
  history: keep_last
  depth: -1
```
2. Run `ros2 bag record -a --qos-profile-overrides-path negative_depth.yaml `

## System (please complete the following information)
 - OS: OSX 10.14, Ubuntu Focal
 - ROS 2 Distro: Foxy pre release binaries

emersonknapp, 2020-07-31:
First person to touch this to double check that it's still happening.

emersonknapp, 2020-07-31:
AWS team picking this up for planned completion by next WG meeting

jikawa-az, 2020-08-01:
> First person to touch this to double check that it's still happening.

Receive the following error when inputting negative values:

`2020-07-31 23:52:07.012 [RTPS_QOS_CHECK Error] INCORRECT TOPIC QOS (rt/talker): depth must be > 0 -> Function checkQos
zsh: segmentation fault  ros2 bag record -a -o my_bag2 --qos-profile-overrides-path `

Afterwards Python crashes.

PR #483 (2020-08-05): Validate QoS profile values are not negative.
Closes #431
closed 2020-08-05 by jikawa-az