ros2/rosbag2 #439: Flake8 Action CI broken

emersonknapp, 2020-06-12:
## Description
The Flake8 linter tests in our Action CI for this package is broken. This happened due to using a version from Eloquent and installing Flake8 via pip, which got us flake8 version 3.8. The release (as of the time of this writing) in Eloquent for ament_lint is not flake8 3.8 compliant.

A fix has already been backported into eloquent, but we should now be running against the latest Foxy builds anyways.

### Expected Behavior
* Flake8 passes on Pull Requests
* We are using ament_lint from ros2-testing for Foxy to run the linter test

### Actual Behavior
Always red

## To Reproduce
Open a Pull Request

## System (please complete the following information)
GitHub

## Additional context

Karsten1987, 2020-07-23:
@emersonknapp I am closing this with #440 being merged. Feel free to reopen.

closed 2020-07-23 by Karsten1987

emersonknapp, 2020-07-23:
It was disabled by #440, fixed upstream, then re-enabled by #467 - just for the context to all be here :)