ros2/rosbag2 #447: Builds prioritizing binaries instead of source

jhdcs, 2020-06-23:
## Description
When attempting to build a rosbag2 project involving an issue I was working on, it seemed that Colcon preferred to use the ROS2 Binary installation rosbag2 files as a source rather than the source code in the project, violating how overlays/underlays are supposed to work. Specifically, rosbag2_cpp was using the unmodified header files from /opt/ros/foxy rather than the modified files in rosbag2_storage.

### Expected Behavior
The modified source files of the project would be used to build rosbag2_cpp, providing the newly-added constant.

### Actual Behavior
The build would use the files from /opt/ros/foxy, which lacked the constant and caused the following error:

`rosbag2_cpp/writers/sequential_writer.cpp: In constructor ‘rosbag2_cpp::writers::SequentialWriter::SequentialWriter(std::unique_ptr<rosbag2_storage::StorageFactoryInterface>, std::shared_ptr<rosbag2_cpp::SerializationFormatConverterFactoryInterface>, std::unique_ptr<rosbag2_storage::MetadataIo>)’:
/home/jhassold/rosbag_ws/src/rosbag2/rosbag2_cpp/src/rosbag2_cpp/writers/sequential_writer.cpp:65:63: error: ‘MAX_BAGFILE_DURATION_NO_SPLIT’ is not a member of ‘rosbag2_storage::storage_interfaces’; did you mean ‘MAX_BAGFILE_SIZE_NO_SPLIT’?
   65 |     std::chrono::seconds(rosbag2_storage::storage_interfaces::MAX_BAGFILE_DURATION_NO_SPLIT)),
      |                                                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                               MAX_BAGFILE_SIZE_NO_SPLIT`

## To Reproduce
1. Install and source the binary installation of Foxy
2. In `rosbag2_storage/storage_interfaces/base_io_interface.hpp`, add a constant
3. In `rosbag2_storage/storage_interfaces/base_io_interface.cpp`, make sure the constant is set to something. 0 is a lovely number.
4. In `rosbag2_cpp/writers/sequential_writer.cpp`, create a dummy function that uses that constant.
5. Try running `colcon build`. You should get an error that the constant you defined is not a member of `rosbag2_storage::storage_interfaces`

## System (please complete the following information)
 - OS: Ubuntu Focal Fossa running on Docker
 - ROS 2 Distro: Foxy

## Additional context
My workaround for the problem was to use `apt remove ros-foxy-rosbag2-storage`. The project built fine after that

emersonknapp, 2020-07-31:
This ROS Answers post is relevant https://answers.ros.org/question/358314/colcon-build-include-path-from-overlay-workspace/

dirk-thomas, 2020-07-31:
I can't reproduce the problem with the steps described in the original ticket.

> This ROS Answers post is relevant https://answers.ros.org/question/358314/colcon-build-include-path-from-overlay-workspace/

This specific case is being resolved by #479 which just declares the necessary target dependencies.

Karsten1987, 2020-08-17:
I am closing this as #479 addressed the issue. Please feel free to re-open this ticket in your case problem still persists.

closed 2020-08-17 by Karsten1987