ros2/rosbag2 #448: SequentialWriter::open failing to create a bag file

Marwan99, 2020-06-28:
## Description
I am trying to use the API to save a message to a bag. 
I am using `rosbag2_cpp::writers::SequentialWriter` to open a bag (which does not exist yet, expecting the API to create it) then write to it.

### Expected Behavior
Creation of a bag file in the URI specified on the use of `SequentialWriter::open`.

### Actual Behavior
No bag file is being created. I am getting the error messages below and the code stops.
```
[rosbag2_storage]: Could not open 'my_bag/my_bag_0' with 'sqlite3'. Error: Failed to setup storage. Error: Could not read-write open database. SQLite error (14): unable to open database file
[ERROR] [1593304125.490127311] [rosbag2_storage]: Could not load/open plugin with storage id 'sqlite3'.
unknown file: Failure C++ exception with description "No storage could be initialized. Abort"
```

## To Reproduce
That's part of the code I am using, `bag->open(...)` is what cause all the error messages:
```c++
rosbag2_cpp::StorageOptions storage_options = rosbag2_cpp::StorageOptions{};
storage_options.uri = "my_bag";
storage_options.storage_id = "sqlite3";

auto bag = std::make_unique<rosbag2_cpp::writers::SequentialWriter>();
bag->open(storage_options, {"rmw_format", "rmw_format"});
```

## System
 - OS: Ubuntu 18.04
 - ROS 2 Distro: Rolling
 - Version: master https://github.com/ros2/rosbag2/commit/a4d49dbda0d121786d4699f4479d5e895df27bfc

## Additional context
I have dug a bit into the code, it seems to me that `SqliteWrapper::SqliteWrapper` is the method that is supposed to create a db3 file in whatever URI passed to. I did some tests and this function can only create db3 files in existing directories and will fail to create a db if the directory doesn't exist.
https://github.com/ros2/rosbag2/blob/a4d49dbda0d121786d4699f4479d5e895df27bfc/rosbag2_storage_default_plugins/src/rosbag2_storage_default_plugins/sqlite/sqlite_wrapper.cpp#L50-L52

I couldn't see a place where non-existent URIs get created(maybe I am missing something). I think the resolution for this problem could be checking the existence of the URI and creating it if it doesn't exist, this can maybe be performed in this function:
https://github.com/ros2/rosbag2/blob/a4d49dbda0d121786d4699f4479d5e895df27bfc/rosbag2_cpp/src/rosbag2_cpp/writers/sequential_writer.cpp#L36

Apologies if my problem analysis is wrong (likely :-)), Please let me know if I need to provide any extra info.
Thanks in advance!

Karsten1987, 2020-06-30:
You've analyzed the problem absolutely correctly. It's currently the `ros2bag` command line interface which is responsible for creating the directory: https://github.com/ros2/rosbag2/blob/master/ros2bag/ros2bag/verb/record.py#L111

I guess by now we have enough functionality in `rcutils/rcpputils` to integrate that work into `rosbag2_cpp`. So thanks a lot for opening the discussion here.

Karsten1987, 2020-07-23:
closed via #450

closed 2020-07-23 by Karsten1987