ros2/rosbag2 #449: How to write data to rosbag2 to use API

henry0572, 2020-06-29:
## Description
It is not possible to write data directly to rosbag2 using ros2 foxy.
I get the same error as below.

[ROS2]How to write data to rosbag2 in ros2 foxy
https://answers.ros.org/question/355764/ros2how-to-write-data-to-rosbag2-in-ros2-foxy/

I was able to load the data based on the following.

How to read data from rosbag2 in ros2
https://stackoverflow.com/questions/60282296/how-to-read-data-from-rosbag2-in-ros2

### Expected Behavior
Writing succeeds.

### Actual Behavior

`writer.write(message)` is what cause all the error messages:
```
terminate called after throwing an instance of 'std::logic_error' what(): basic_string::_M_construct null not valid
```

## System (please complete the following information)
 - OS: Ubunutu 20.04
 - ROS 2 Distro: Foxy
 - Version: release

emersonknapp, 2020-07-10:
@henry0572 thanks for bringing this up. The C++ API is not straightforward. You can see in @Karsten1987 's PR ^ linked above this comment, a project to reduce the complexity of using the C++ API directly. It also acts as a good example of how to use it as it is currently.

Karsten1987, 2020-07-23:
The following test should illustrate how to use the C++ API as of now. There will be further modifications done which allows users to directly write non-serialized messages in the future. 
https://github.com/ros2/rosbag2/blob/master/rosbag2_tests/test/rosbag2_tests/test_rosbag2_cpp_api.cpp

I'll go ahead and close this issue as it's possible to write data to rosbag2 via the C++ API.

closed 2020-07-23 by Karsten1987