ros2/rosbag2 #453: rosbag record issue

yangqun-yangrong, 2020-07-07:
I record realsense t265 and d435 topics with [ros2 bag record -a].
In the bag files, the image frequency is reduced to around 1.
Is it related to record too many topics? Or anything else?

When I play the bag, I can get all topics with [ros2 topic list], but when I use [ros2 bag record -a] to record the topics from the bag, I could not get all topics.Why?

I do with Jetson nano with ros2_dashing.Thank you.

adamdbrw, 2020-07-07:
You might be running into performance issues. Check #435 and attached issues. You can wait for fixes or improve performance by changing pragma settings in sqlite_writer.

Karsten1987, 2020-07-07:
What are the QoS settings for the realsense data you're trying to record?
Dashing doesn't have the option to record non-reliable data. Can you try Foxy and see if the problem persists?

emersonknapp, 2020-07-31:
another option in Dashing would be to have the cameras publish with a Reliable QoS (though this is not ideal)

I'm going to close this issue for now, please reopen if we can help resolve anything further.

closed 2020-07-31 by emersonknapp