ros2/rosbag2 #454: ros2bag QOS test is failing on windows

Karsten1987, 2020-07-08:
Test is "reliably" failing on windows:
https://ci.ros2.org/view/nightly/job/nightly_win_rel/1605/testReport/ros2bag.test/test_record_qos_profiles/test_record_qos_profiles/

```
FAIL: test_record_qos_profiles.TestRos2BagRecord.test_incomplete_qos_profile
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Traceback (most recent call last):
  File "c:\python38\lib\shutil.py", line 613, in _rmtree_unsafe
    os.unlink(fullname)
PermissionError: [WinError 32] The process cannot access the file because it is being used by another process: 'C:\\Users\\ContainerAdministrator\\AppData\\Local\\Temp\\tmphur3j5pi\\ros2bag_test_incomplete_0.db3'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\python38\lib\tempfile.py", line 803, in onerror
    _os.unlink(path)
PermissionError: [WinError 32] The process cannot access the file because it is being used by another process: 'C:\\Users\\ContainerAdministrator\\AppData\\Local\\Temp\\tmphur3j5pi\\ros2bag_test_incomplete_0.db3'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\ci\ws\src\ros2\rosbag2\ros2bag\test\test_record_qos_profiles.py", line 87, in test_incomplete_qos_profile
    assert not matches, print('ros2bag CLI did not produce the expected output')
```

@emersonknapp do you have an idea?

emersonknapp, 2020-07-09:
I don't know immediately but will begin investigating

dirk-thomas, 2020-07-14:
@emersonknapp Any update on this?

dirk-thomas, 2020-07-15:
Please see #462 which modifies the test to only create one temp dir for all tests within the test class. That allows it to be cleanly removed on Windows.

dirk-thomas, 2020-07-17:
The test is still failing (even though it passed CI on the PR) - reopening...

emersonknapp, 2020-07-17:
Noted - we are pulling this in as top priority for Monday

jaisontj, 2020-09-22:
@dirk-thomas Could you take a look at: #525 ? It gets rid of most of the `time.sleep` except for the one inside `tearDownClass`. Was not able to find a better way around it. However, given the title of this issue, we no longer use several sleeps in the test, can this issue be closed?

dirk-thomas, 2020-09-22:
> can this issue be closed?

Sounds good to me. Once the PR is merged that should happen automatically.

PR #462 (2020-07-15): use a single temp dir for the test class
Fixes #454: https://ci.ros2.org/view/nightly/job/nightly_win_rel/1613/testReport/ros2bag.test/test_record_qos_profiles/test_record_qos_profiles/

[image]

The diff ignoring whitespace changes is much easier to read: https://github.com/ros2/rosbag2/pull/462/files?w=1
PR #525 (2020-09-22): [ros2bag test_record] Gets rid of time.sleep and move to using command.wait_for_output
Tries to fix #454 

Looks to me that the original problem described at #454 was probably addressed by #466 
However, before #466, there was #462 which moved from using different temp directory per test to using one temp directory for all the tests - this PR also introduced the `time.sleep` within each test and also one within the cleanup block (which deletes the temp directory)

In this PR:
- I've moved from using `time.sleep` within each tests to using `bag_command.wait_for_output` which essentially waits f [... cut]
closed 2020-09-22 by jaisontj