ros2/rosbag2 #458: Troubleshooting link needs to be updated

togaen, 2020-07-13:
## Description
In rosbag2_transport the `__init__.py` there is a link to the troubleshooting docs in case import fails. However, the link no longer works. The current link is:

```
https://index.ros.org/doc/ros2/Troubleshooting/#import-failing-even-with-library-present-on-the-system
```

I believe it should be:

```
https://index.ros.org/doc/ros2/Troubleshooting/Installation-Troubleshooting/#import-failing-even-with-library-present-on-the-system
```

### Expected Behavior
The link should go to a valid destination.

### Actual Behavior
The link ends up loading a generic landing page.

## To Reproduce
Visit the link in the code.

## System (please complete the following information)
 - OS: Ubuntu 20.04
 - ROS 2 Distro: Foxy
 - Version: (from apt) 0.3.3-1focal.20200708.064112

## Additional context
** Add any other context about the problem here **

Karsten1987, 2020-07-13:
Thanks for reporting. The issue was already reported in https://github.com/ros2/rosbag2/issues/445
Closing this as duplicate.

closed 2020-07-13 by Karsten1987

togaen, 2020-07-13:
Sorry! I should have done a better job searching the issues first.