ros2/rosbag2 #469: New Test failures in Rci Nightly

mjcarroll, 2020-07-23:
## Description

There are new failing tests in the Rolling nightly CI job. http://build.ros2.org/view/Rci/job/Rci__nightly-debug_ubuntu_focal_amd64/21/consoleFull#console-section-1942

While all of the actual tests pass, at the end of execution there is a LeakSanitizer error:

```
04:58:11 6: ==3418==LeakSanitizer has encountered a fatal error.
04:58:11 6: ==3418==HINT: For debugging, try setting environment variable LSAN_OPTIONS=verbosity=1:log_threads=1
04:58:11 6: ==3418==HINT: LeakSanitizer does not work under ptrace (strace, gdb, etc)
04:58:11 6: -- run_test.py: return code 23
04:58:11 6: -- run_test.py: inject classname prefix into gtest result file '/tmp/ws/test_results/rosbag2_cpp/test_ros2_message.gtest.xml'
04:58:11 6: -- run_test.py: verify result file '/tmp/ws/test_results/rosbag2_cpp/test_ros2_message.gtest.xml'
04:58:11 1/1 Test #6: test_ros2_message ................***Failed    0.12 sec
```

This happens in both debug and release variants.

### Expected Behavior

The tests pass in the CI environment.

### Actual Behavior

The tests fail and cause the nightly job to fail.

## System (please complete the following information)
 - OS: Ubuntu Focal
 - ROS 2 Distro: Rolling
 - Version: Rolling

emersonknapp, 2020-07-23:
Thanks for reporting, I've moved it to the top of the bug queue

emersonknapp, 2020-07-23:
Thanks for reporting this - we will put it on the top of our input queue

dirk-thomas, 2020-07-27:
The test failure has been resolved in #472 with #454 remaining.

closed 2020-07-27 by dirk-thomas