ros2/rosbag2 #474: ros2 bag record(from foxy) not able to record the topics(from dashing)

pulkitgoyal1506, 2020-07-28:
## Description
In my setup, I have two PCs on a LAN. First PC has the ROS (version dashing) running on it with topics publishing some data. Second PC has a docker container with foxy ros version running on it. I am recording data published on the topics of first PC running ros dashing, using rosbag recorder from ros foxy in the docker on second PC.

I am following this tutorial for recording data on ros foxy.
https://index.ros.org/doc/ros2/Tutorials/Ros2bag/Overriding-QoS-Policies-For-Recording-And-Playback/#id3
(Reason for this setup: As the dashing bag recorder cannot support the recording of data with different QoS settings, I am trying to record the dashing data using the foxy bag recorder.)

### Expected Behavior
I expect the foxy bag recorder to record the data from dashing topics.

### Actual Behavior
Foxy bag recorder is not recording any data coming from dashing topics. Executing 'ros2 topic list' on docker with foxy installed is listing all the topics on the dashing side. But 'ros2 bag record' is not subscribing to any of the topics.

Output of ros2 bag record command:
[image]

## To Reproduce

- PC1 with ROS2 dashing publishing data on some topics
- PC2 with docker having ROS2 foxy recording that data

## System (please complete the following information)
- PC1: ROS 2 Dashing, Ubuntu 18.04
- PC2: Docker with ROS 2 Foxy

## Additional context

emersonknapp, 2020-07-29:
This is a consequence of the underlying ROS 2 communication infrastructure. Rosbag2 just uses subscriptions to receive messages, so it's limited to that behavior. I think that this would be better filed against github.com/ros2/ros2 - I don't think there's anything that can be done about it here.

pulkitgoyal1506, 2020-07-29:
Thanks for quick response. I have raised an issue on ros2 repository, closing it here.

closed 2020-07-29 by pulkitgoyal1506