ros2/rosbag2 #477: Option to log whenever a message is known to be dropped

emersonknapp, 2020-07-31:
## Description

The `rosbag2` writer should log via the CLI whenever a message (compressed or uncompressed) has been dropped and not written to the storage plugin. 

## Test Plan

* Unit test demonstrating writer logs the warning if the message wasn't written to the database (mocking the storage plugin).
* Unit test demonstrating writer logs the warning in a compression workflow (i.e. the compression flag was passed to the write).

## Documentation Plan

Code docstrings

## Acceptance Criteria

* [ ] Writer can log on dropped messages

dabonnie, 2020-08-06:
Wouldn't it be useful to keep a total count dropped? This way a percentage dropped could be displayed at the end of recording.

emersonknapp, 2020-08-07:
Yeah, I think that would be useful

adamdbrw, 2020-10-19:
Currently in high throughput scenarios, messages are dropped due to bottleneck in I/O (and its handling in rosbag2). The issue propagates all the way back to the callback queue, which isn't processed fast enough and it fills / drops messages. Thus, the core part of lost message counting would have to include what happens in the callback queue. What are the mechanisms currently to monitor that?

emersonknapp, 2020-10-19:
The callback queue for the subscriptions themselves? I'm not sure that there is currently a mechanism to monitor that - it may require a new RMW API to expose the information

adamdbrw, 2021-01-22:
Currently there is one way that the dropped messages are logged. On the rosbag2 side we can only be aware of messages that were successfully delivered by transport and managed to not get overwritten in a subscriber queue due to a limit. Unfortunately, we don't have a mechanism now to count whether the message was overwritten in a subscription queue. However, due to producer-consumer scheme currently in place for the writer (double buffering in cache) messages lost due to performance are lost in cache (if the consumer cannot keep up in clearing the queue). This loss is currently logged at exit (summary of loss per topic).

Is this sufficient? We could also add this information to the metadata.yaml itself, although it does not pertain to the recorded data itself.

y-okumura-isp, 2021-03-05:
We encouned the similar situation, so I want to share.
In our environment, the bottleneck looks topic communication rather than rosbag record.

Here is our setup. We used rosbag2_performance_benchmarking.

- Spec
  - Intel(R) Core(TM) i7-4790 CPU @ 3.60GHz
  - 32 GB RAM
  - HDD: TOSHIBA MD05ACA8(SATA)
  - ROS2 Rolling on docker(osrf/ros2:devel based image)

- Condition
  - 10 KB, 100 Hz = 1 MB/s per 1 publisher.
  - `msg_count_each`: `24000`
  - we changed load by `publisher_count`.

In no transport mode, by changing `max_cache_size` 500MB, we could record 100 `publisher_count` successfully.
We choose 500MB because the sqlite insertion method blocked a few seconds in our environment.
I think more publishers are OK by adjusting `max_cache_size`.

In transport mode, we started with the condition above: 500 MB `max_cache_size` 500MB and 100 `publisher_count`.
The rosbag2 record could save all the messages it recieved.
But when we counted how many times each subscription callbacks are called, we found some callbacks dropped.
For example, callback of `/rosbag2_performance_benchmarking_node/benchmarking_small_8` was called 23998 times, not 24000 times.
Even if we set QoS as KeepLast with 24000 qos depth and Reliable, the situation was the same.

I don't know such a high frequecy is common or not.
To add, the goal of our experiment was only to search the limitation.
But I feel happy if I could know whether (and where, when) the drop happened.

MichaelOrlov, 2025-03-27:
- I'm closing this issue since we already have a notification about the number of messages dropped from the cache buffers due to the slow IO at the end of the recording.

- The more throw statistics about messages dropped will be developed as a separate feature.

closed 2025-03-27 by MichaelOrlov