ros2/rosbag2 #484: Random Access Reader

nslay, 2020-08-01:
## Description

The original rosbag API provided a View of a rosbag that would provide C++ iterators to MessageInstances. Iterators states could be copied and access messages anywhere in the bag. The rosbag2 API only seemingly supports a sequential reading operation (read_next()). This is extremely limited compared to the original rosbag API.

Application? The original rosbag API can be used to implement advanced playback features not available in 'rosbag play'. These are features that are normally found in video players like mpv or VLC. Features related to random access bag iterators include interactive seeking in bags (forward and backward), interactive message stepping (forward and backward), interactive AB loop points, etc... These features may seem odd for ROS, but they facilitate considerably accelerated viewing of rosbags and debugging of ROS pipelines. The current rosbag2 API is too limited to implement these playback features save for buffering every single message across possibly multiple bags used in playback. The latter is certainly possible, but extraordinarily inefficient.

## Related Issues

** Is this feature dependent on any other features? Is it part of a larger project? Note here. **

## Completion Criteria

The rosbag2 API (notably Reader and BaseReaderInterface) needs to expose a seekable interface. Either something like rosbag's View iterators or something like low-level OS file seeking facilities.

## Implementation Notes / Suggestions

The BaseReaderInterace should have something like a function like seek() and something like has_seek() to determine if seeking is possible, and something like tellg() (name from C++'s fstream) to get the current message position in the bag. Additionally providing a function that produces the receipt timestamp of the current message would help immensely.

I imagine there would be an implementation of the BaseReaderInterface interface called RandomAccessReader like we currently have with SequentialReader. Or perhaps RandomAccessReader can extend SequentialReader.

The Reader interface itself should forward these functions (seek(), has_seek() and tellg()).

## Testing Notes / Suggestions

A good and simple test could be looping a subset of the messages in a bag. This requires seeking somewhere in the middle of the bag (loop point A) and iterating the messages (to loop point B) and repeating this.

emersonknapp, 2020-08-14:
Would be nice to see a design proposal before someone starts implementing this - but the feature request is clear

nslay, 2020-08-14:
I have a full featured player for ROS1 that can do this fancy stuff. I'm pushing to get it open sourced which could provide a clearer idea of what is needed and how rosbag2 might be designed. I can't promise I'll succeed in getting it open sourced though :(

Kaju-Bubanja, 2021-01-26:
@nslay Any success in open sourcing the player? Sounds like this would be tremendously useful. I think this is the holy grail of the ros bag api. I also think this feature request is very useful. In case @nslay can not open source his implementation, can somebody more senior provide some guidance for how to implement this? Since I really need this feature I will be investing some time into it. I just don't want to go down the wrong rabbit hole design wise. Also how does this feature request interplay with [PR 443](https://github.com/ros2/rosbag2/pull/443)? I guess this Issue is an superset of the pause feature. But since it seems like the pause feature is close to completion, I am wondering if some work done on that PR can be used as a starting point for this one? Or if this needs to be done completely different.

emersonknapp, 2021-01-26:
> Also how does this feature request interplay with [PR 443](https://github.com/ros2/rosbag2/pull/443)? I guess this Issue is an superset of the pause feature

Sort of - it depends how you want to define the feature set. The pause feature is specifically part of the Player, to allow its playback loop to be interrupted and resumed, plus it will also eventually have keyboard interaction as a UI to interface with that feature.

The first and most useful thing for "Random access rosbag2" would be an API update to the Reader, so that it has more options than "has_next" and "read_next", which are the only things it can do right now. Integrating any new features into a commandline-usable Player would be a secondary consideration, the important part would be exposing the functionality as an API so that tools could be built around it

nslay, 2021-01-27:
No luck open sourcing it. I've since left the robotics community and went back into medical imaging AI.

I think I can at least offer a big picture of the approach to writing a player with advanced playback features matching that of mplayer/mpv/vlc.

The rosbag API (v1) provides an iterator that can be incremented, copied and dereferenced at any time. So you can have several iterators to different messages in an open bag and be able to access the message contents by dereferencing the iterators. This is the key to pulling this off!

In the simplest case, with playback with 1 bag, what you can do is iterate a bag and keep checkpoints (e,g. say 5 seconds) of message iterators. Then it's easy to "snap" to the nearest message when doing something like seeking. You just do something clever like a binary search to find the nearest message iterator to the requested time in your buffer of checkpoints. Then you just resume playback from that iterator.

You might think it will be hideously complicated with multiple bags with overlapping/disjoint time ranges. In fact, it's not much more complicated! When you have multiple bags opened with iterators, you need to be able to pick the next newest iterator among all the bags, right? That's just a priority queue. So you make a priority queue of bags/iterators. You pop an iterator off the top, increment it... if it's not at the end, push it back in the queue.

Now instead of checkpointing individual iterators for seeking... you checkpoint priority queue states. When you seek, you take the priority queue nearest to the requested seek time and you just resume playback from that priority queue state. The priority queues are assumed lightweight (because they're basically just holding iterators/pointers).

Now what about /clock (if it's enabled)? You can just make like a pretend message iterator that coexists in the priority queue with the bag iterators. So if you do anything like seeking, /clock is seamlessly handled for you!

What about finer message stepping (forward or backward)? You do something similar to the checkpointing buffer of priority queues. You keep a small history of consecutive states of priority queues. Now _consecutive_ can mean more than just the individual messages in order (which isn't very useful). You can instead think of only messages in order that match a topic pattern or messages that match a type or something like that (specified by the user). So if somebody wants to step a "message/frame" forward/backward... you just look in this little localized buffer of priority queues to do that (in mpv, this would be period [.] for one "frame" forward and comma [,] for one "frame" backward). You know, just keep something small like 20 seconds of these priority queue states (centered so that you have 10 seconds in past and 10 seconds in future). That should be plenty for someone wanting to step back like a few messages/frames if they happened to miss something!

That's the basics anyway. There are so many more details that I am missing! The UI aspects... the little things like how you detect and deal with large gaps of time between messages (remember /clock might share the queue!)? Maybe latched topics (yuck!)? And you probably want to do some multi-threading (like have one dedicated thread that constructs and maintains the checkpointing buffer and another thread that does the playback). Getting the playback timing to be correct amidst users interactively adjusting playback speeds or seeking or stepping frames... or whatever... is slightly challenging!

Oh, and I once complained on ros_common about ros::Time throwing exceptions when no other timestamp scheme in Unix or Windows behaves this way (time_t, timespec, timeval, macros/functions that manipulate these, etc...!). If you implement this player, you may come to hate ros::Time... and you may find yourself often converting all your timestamps to doubles (toSec()) to avoid exceptions from crashing your fancy player. Negative timestamps can be useful and I think it was foolish to put exceptions into ros::Time. And you may feel that pain when writing a player that does this kind of crazy interactive playback. IMO, don't throw exceptions for simple things like time stamps! Somebody may want or experience negative timestamps in the normal course of computation (and those programmers will check for it)! I think ros::Time is great... it's too bad I often avoided directly using it because of your exception handling you foisted on us for some strict correctness (that I didn't want here).

emersonknapp, 2021-08-06:
Related to #821

nslay, 2023-03-29:
Would it help if I implemented a new ROS1 player from scratch that has seek ability? I think it's been enough years... can hardly remember anything.

MichaelOrlov, 2023-03-30:
- Closing this issue as done. 
Since we have added player-seek(timestamp) API
https://github.com/ros2/rosbag2/blob/d926a617c12831b422f4eb364e5e886aec988461/rosbag2_transport/src/rosbag2_transport/player.cpp#L440
and even corresponding seek service request 
https://github.com/ros2/rosbag2/blob/d926a617c12831b422f4eb364e5e886aec988461/rosbag2_interfaces/srv/Seek.srv#L1-L3

closed 2023-03-30 by MichaelOrlov