ros2/rosbag2 #485: Playback on macOS

Stapelzeiger, 2020-08-02:
## Description
Playing back a bag that contains only the *.db3 file (without the *.db3-shm, *.db3-wal) fails on macOS with `"SQLite error (14): unable to open database file"`

### Expected Behavior
Bag plays correctly.

### Actual Behavior
fails to play with error message described above.

## To Reproduce
open a bag that doesn't contain *.db3-shm/*.db3-wal files on macOS:
`ros2 bag play <name-of-bag>`

to get such a bag either create it on aarch64 ubuntu (those don't contain the files for some reason)
or open the .db3 in https://sqlitebrowser.org and close it. This removes the *.db3-shm/*.db3-wal files.

## System
 - OS: macOS 10.14
 - ROS 2 Distro: Foxy
 - Version: binary patch 1

## Additional context

The problem error comes from [this line](https://github.com/ros2/rosbag2/blob/6ff9d61535c59ee539d0f70f1e1a017f467254d6/rosbag2_storage_default_plugins/src/rosbag2_storage_default_plugins/sqlite/sqlite_wrapper.cpp#L48).
From what I understand, the database is in WAL mode and the *.db3-shm, *.db3-wal have to be created if they don't exist. This seems to fail when opening in read-only mode for some reason.

Changing to `SQLITE_OPEN_READWRITE` [in the sqlite3_open_v2](https://github.com/ros2/rosbag2/blob/6ff9d61535c59ee539d0f70f1e1a017f467254d6/rosbag2_storage_default_plugins/src/rosbag2_storage_default_plugins/sqlite/sqlite_wrapper.cpp#L40) fixes the error, the extra files are created and the bag plays correctly.

I believe this fixes half of the issue #153 (the playback of an armhf bag on macOS), but I haven't tested it yet.

Karsten1987, 2020-08-04:
very interesting. Thanks for reporting this. Do you mind opening a PR with your proposed fix?

Stapelzeiger, 2020-08-04:
Sure, I'll do that.

MichaelOrlov, 2022-08-31:
@Stapelzeiger At some point we changed `PRAGMA journal_mode=MEMORY` by default instead of `PRAGMA journal_mode=WAL`.
Therefore database not in `WAL` mode by default any more. 
Therein this issue shouldn't be reproducible any more.  
However I don't have macOS to reproduce or test it.

I am closing this issue as stale and fixed via workaround. Feel free to reopen it if issue still reproduces.

closed 2022-08-31 by MichaelOrlov