ros2/rosbag2 #486: Database access race condition when recording

Stapelzeiger, 2020-08-04:
## Description

Recording rosbags with lots of topics and high message rate causes race condidtions when writing to the database.
I believe the problem is that new topic entries are written while at the same time message data from other topics is coming in.

### Expected Behavior
`ros2 bag record -a` records all topics without errors

### Actual Behavior

The problem manifests itself in about 1 in 10 runs (with the launchfile below) in one of the following three ways:
- recording seems fine, then right after hitting ctrl-C the it logs an error and exits, the bag is almost empty (I believe something gets stuck in the topic setup)
```
...
^C[INFO] [1596518181.782926826] [rclcpp]: signal_handler(signal_value=2)
[ERROR] [1596518182.245972174] [rosbag2_transport]: Failed to record: Error when processing SQL statement. SQLite error (5): database is locked
```
- segfault a few seconds after recording starts, while still adding new topics
- error "database disk image is malformed" a few seconds after recording starts, again while adding topics (this seems to be the most common)
```
...
[INFO] [1596518141.207651288] [rosbag2_transport]: Subscribed to topic '/topic57'
[ERROR] [1596518141.560762687] [rosbag2_transport]: Failed to record: Error when processing SQL statement. SQLite error (11): database disk image is malformed
```

## To Reproduce

run many publishers. The launch file below seems to work, although not as well as my robot which is publishing at higher data rate (but fewer topics).
```
from launch import LaunchDescription
from launch_ros.actions import Node

def generate_launch_description():
    return LaunchDescription([
        Node(
            package='examples_rclcpp_minimal_publisher',
            executable='publisher_lambda',
            name='pub'+str(i),
            remappings=[
                ('/topic', '/topic'+str(i))
            ]
        ) for i in range(70)
    ])
```
then run `ros2 bag record -a`

## System (please complete the following information)
- OS: macOS 10.14, ubuntu 20.4 (aarch64)
- ROS 2 Distro: Foxy
- Version: binary patch 1

## Additional context

[Opening the database](https://github.com/ros2/rosbag2/blob/6ff9d61535c59ee539d0f70f1e1a017f467254d6/rosbag2_storage_default_plugins/src/rosbag2_storage_default_plugins/sqlite/sqlite_wrapper.cpp#L52) in `SQLITE_OPEN_FULLMUTEX` mode instead of `SQLITE_OPEN_NOMUTEX ` fixes this. I've also tried manually adding a lock around the topic entry write and the messages write, which seems to work as well.

See also #480 for a related issue. (`database_->get_last_insert_id()` is sometimes incorrect, also caused by multiple threads using the database)

Stapelzeiger, 2020-08-05:
relevant sqlite documentation: https://www.sqlite.org/threadsafe.html

> Multi-thread. In this mode, SQLite can be safely used by multiple threads provided that no single database connection is used simultaneously in two or more threads.
> Serialized. In serialized mode, SQLite can be safely used by multiple threads with no restriction.
> ...
> The SQLITE_OPEN_NOMUTEX flag causes the database connection to be in the multi-thread mode and the SQLITE_OPEN_FULLMUTEX flag causes the connection to be in serialized mode. 

I think the bigger question is whether the storage plugins are supposed to be thread safe. The recorder accesses the storage from multiple threads [here](https://github.com/ros2/rosbag2/blob/d810a4ceaa3c85cf821ac20fb14c7ed15ec2b79a/rosbag2_transport/src/rosbag2_transport/recorder.cpp#L70) (launching the discovery and in parallel receiving message callbacks from ros).

However I don't think the sqlite storage plugin has been written with thread-safety in mind. (for example the `std::unordered_map<std::string, int> topics_` is used in both `create_topic()` and `write()` without mutex). This means simply switching to `SQLITE_OPEN_FULLMUTEX` is not enough to ensure thread safety.

Karsten1987, 2020-08-05:
I am certain that we didn't develop the storage plugin with thread-safety in mind. Your finding is absolute correct and it should use a lock when discovery is enabled.

Stapelzeiger, 2020-08-06:
Great, then we've found the problem. I opened a PR that fixes the recorder.

PR #491 (2020-10-21): Mutex around writer access in recorder
The storage plugins are not designed to be thread safe. The recorder must therefore use a mutex when accessing the writer from different contexts. This happens for topic creation during discovery as well as in message callbacks from subscribed topics.

This fixes #486 and #480.
closed 2020-10-21 by Karsten1987