ros2/rosbag2 #499: rosbag2 memory leak

dawonn-haval, 2020-08-08:
## Description
I was running `my_test_plugin` with a debug trace as described in #498 overnight; when I returned in the morning the node was killed by the kernel for consuming all of the memory in the system.

[image]

### Expected Behavior
No memory leaks.

## System
 - OS: U20.04
 - ROS 2 Distro: Foxy
 - Version: master (f1d1145de6762)

## Additional context
I was attempting to track down why lidar and image topics stop (and apparently restart) recording. Here is the final output of the trace from #498:
[image]

Stapelzeiger, 2020-08-10:
What RMW are you using?

dawonn, 2020-08-10:
Default: fastrtps

dawonn-haval, 2020-08-10:
Perhaps interesting note: if I run rosbag2 in valgrind I never receive any messages from the 'large' topics. If I stop and run without valgrind, i do receive them.

dawonn-haval, 2020-08-10:
https://github.com/ros2/rosbag2/blob/f1d1145de6762ec0d3d19246f7a374ffdd4441fe/rosbag2_transport/src/rosbag2_transport/recorder.cpp#L170

It's not so much that rcl expects the client to deconstruct the message, but rosbag2 explicitly takes ownership of the data there. I'm still trying to figure out how the destructor is called. I expected to see rmw_serialized_message_fini() be called, but don't see any references to it. Might be on to something?

Karsten1987, 2020-08-10:
I could see that there's indeed a bug introduced in [this PR](https://github.com/ros2/rosbag2/pull/386/files). 
As @dawonn pointed out we transfer ownership via `release_rcl_serialized_message()`, but the bag message doesn't clean up the serialized message. 
The likely fix for this is to replace the `std::make_shared` call with a instantiation of that shared pointer with a custom destructor which cleans up the buffer.

Karsten1987, 2020-08-10:
see https://github.com/ros2/rosbag2/pull/502 for a potential fix. Any feedback on the PR is welcome.

Karsten1987, 2020-08-19:
closed via #502

closed 2020-08-19 by Karsten1987