ros2/rosbag2 #501: rosbag2_transport test_record failing

dirk-thomas, 2020-08-10:
Similar to #138,

See recent nightlies:
* https://ci.ros2.org/view/nightly/job/nightly_linux-aarch64_repeated/1286/testReport/(root)/projectroot/test_record__rmw_fastrtps_cpp/
* https://ci.ros2.org/view/nightly/job/nightly_osx_repeated/2058/testReport/(root)/projectroot/test_record__rmw_fastrtps_cpp/
* https://ci.ros2.org/view/nightly/job/nightly_linux-centos_repeated/490/testReport/(root)/projectroot/test_record__rmw_cyclonedds_cpp/

@Karsten1987 @emersonknapp Please look into this soon since it makes the nightly builds unstable.

chapulina, 2020-11-02:
Still flaky:

* CycloneDDS: https://ci.ros2.org/view/nightly/job/nightly_linux-aarch64_repeated/1396/testReport/(root)/projectroot/test_record__rmw_cyclonedds_cpp/history/
* FastRTPS: https://ci.ros2.org/view/nightly/job/nightly_linux-aarch64_repeated/1396/testReport/(root)/projectroot/test_record__rmw_fastrtps_cpp/history/

In both cases, the problem seems to be localized into a specific test case, so maybe marking the entire test executable as `xfail` may be too invasive. Maybe the maintainers could consider prepending the offending test case with `DISABLED_` to skip just that one.

```
[ RUN      ] RecordIntegrationTestFixture.receives_latched_messages
[INFO] [1604315840.494966010] [rosbag2_transport]: Listening for topics...
[INFO] [1604315840.497894250] [rosbag2_transport]: Subscribed to topic '/latched_topic'
[INFO] [1604315840.499024942] [rosbag2_transport]: All requested topics are subscribed. Stopping discovery...
free(): invalid next size (fast)
-- run_test.py: return code -6
-- run_test.py: generate result file '/home/jenkins-agent/workspace/nightly_linux-aarch64_repeated/ws/build/rosbag2_transport/test_results/rosbag2_transport/test_record__rmw_cyclonedds_cpp.gtest.xml' with failed test
-- run_test.py: verify result file '/home/jenkins-agent/workspace/nightly_linux-aarch64_repeated/ws/build/rosbag2_transport/test_results/rosbag2_transport/test_record__rmw_cyclonedds_cpp.gtest.xml'
```

clalancette, 2020-11-17:
The last time that this failed was Nov 2 (the date of the last comment).  I don't have anything definitive to point to that solved it, but it does seem to have improved.  I'm going to close this out for now, but if it pops back up please feel free to reopen.

closed 2020-11-17 by clalancette