ros2/rosbag2 #521: Ros2 bag "Error reading SQL query"

salmagro, 2020-09-07:
## Description
Dear community, I am trying to replay a ros2 bag but I am getting a SQLite error. Files included in folder:
```
-rwxrwxrwx 1        44749 Sep  4 18:22 metadata.yaml
-rwxrwxrwx 1  23821316096 Sep  4 18:22 rosbag2_2020_09_04-17_28_07_0.db3
-rwxrwxrwx 1        32768 Sep  7 13:43 rosbag2_2020_09_04-17_28_07_0.db3-shm
-rwxrwxrwx 1            0 Sep  7 09:49 rosbag2_2020_09_04-17_28_07_0.db3-wal
```

### Expected Behavior
Replay a ros2 bag.

### Actual Behavior
```
[INFO] [1599479010.313701282] [rosbag2_storage]: Opened database 'rosbag2_2020_09_04-17_28_07//rosbag2_2020_09_04-17_28_07_0.db3' for READ_ONLY.
[ERROR] [1599479010.534636663] [rosbag2_transport]: Failed to play: Error reading SQL query. SQLite error (11): database disk image is malformed
```

## To Reproduce
Running the command for replaying a saved bag.
`ros2 bag play rosbag2_2020_09_04-17_28_07/`

## System (please complete the following information)
 - OS: Debian 19
 - ROS 2 Distro: Foxy

dabonnie, 2020-09-08:
Hi @salmagro, thanks for filing this issue. I have a few question for you:

* what was the command used to record data?
* how was the recording stopped?
* does `ros2 bag info rosbag2_2020_09_04-17_28_07/rosbag2_2020_09_04-17_28_07_0.db3` provide any data?
* have you tried to record and playback another bag file? 
* could you please clarify your operating system version? [ros2 supports Debian 10](https://www.ros.org/reps/rep-2000.html#foxy-fitzroy-may-2020-may-2023) and I'm unaware of [Debian 19](https://www.debian.org/releases/)

salmagro, 2020-09-08:
Hi @dabonnie Thank you very much for your answer. 
* Command used: $ ros2 bag record -a
* Recording stopped with: crtl + c
* Yes provide all topics recorded..
* Yes but they present the same behaviour.
* Sorry it was a typo. It is Debian 10.

dabonnie, 2020-09-08:
@salmagro thanks for the info. I'm will attempt to reproduce your issue on a Debian 10 system.

dabonnie, 2020-09-08:
@salmagro how did you install ROS2? FWIW after some simple testing (via building from source) I was able to record and playback a simple bagfile on Debian 10. 

Data generation: ` ros2 topic pub /chatter std_msgs/String "data: Hello world"`
Record command: `ros2 bag record -a`

Info and playback commands worked as expected. Can go down the path of testing the binaries as well or attempting to record with higher bandwidth inputs (if you can provide some details that would be helpful).

dabonnie, 2020-09-08:
Another approach we can take is can you open a '*.db3' file with the [SQLite CLI tool](https://sqlite.org/cli.html)? After opening it can you immediately close it?

```
While a database connection is open on a WAL-mode database, SQLite maintains an extra journal file called a "Write Ahead Log" or "WAL File". The name of this file on disk is usually the name of the database file with an extra "-wal" suffix, though different naming rules may apply if SQLite is compiled with SQLITE_ENABLE_8_3_NAMES.

The WAL file exists for as long as any database connection has the database open. Usually, the WAL file is deleted automatically when the last connection to the database closes. However, if the last process to have the database open exits without cleanly shutting down the database connection, or if the SQLITE_FCNTL_PERSIST_WAL file control is used, then the WAL file might be retained on disk after all connections to the database have been closed. The WAL file is part of the persistent state of the database and should be kept with the database if the database is copied or moved. If a database file is separated from its WAL file, then transactions that were previously committed to the database might be lost, or the database file might become corrupted. The only safe way to remove a WAL file is to open the database file using one of the sqlite3_open() interfaces then immediately close the database using sqlite3_close().
```

From Section 4: https://sqlite.org/wal.html

salmagro, 2020-09-09:
Thank you very much for your help, I have applied the commands for opening, exporting and importing again the data base mentioned in:

*11. Converting An Entire Database To An ASCII Text File* https://sqlite.org/cli.html

I was able to replay again de big data base. 
What will be the right way to stop the record of a rosbag?

closed 2020-09-09 by salmagro

dabonnie, 2020-09-09:
> Thank you very much for your help, I have applied the commands for opening, exporting and importing again the data base mentioned in:
>
> _11. Converting An Entire Database To An ASCII Text File_ [...]

Glad this workaround was successful! You should be able to exit cleanly with SIGINT (CTRL+C), which I verified myself on Debian 10. If you continue to see this behavior please let us know and reopen the issue.