ros2/rosbag2 #522: rosbag fails at importing import_c_library

christianrauch, 2020-09-10:
## Description
ros2 bag fails to record or play bag files

### Expected Behavior
ros2 bag can record and play

### Actual Behavior
I tried to record on ROS2 foxy via `ros2 bag record <log-file-name>` which fails with:
```
Traceback (most recent call last):
  File "/opt/ros/foxy/bin/ros2", line 11, in <module>
    load_entry_point('ros2cli==0.9.7', 'console_scripts', 'ros2')()
  File "/opt/ros/foxy/lib/python3.8/site-packages/ros2cli/cli.py", line 67, in main
    rc = extension.main(parser=parser, args=args)
  File "/opt/ros/foxy/lib/python3.8/site-packages/ros2bag/command/bag.py", line 38, in main
    return extension.main(args=args)
  File "/opt/ros/foxy/lib/python3.8/site-packages/ros2bag/verb/record.py", line 141, in main
    from rosbag2_transport import rosbag2_transport_py
  File "/opt/ros/foxy/lib/python3.8/site-packages/rosbag2_transport/__init__.py", line 15, in <module>
    from rpyutils import import_c_library
ImportError: cannot import name 'import_c_library' from 'rpyutils' (/opt/ros/foxy/lib/python3.8/site-packages/rpyutils/__init__.py)
```

## To Reproduce
1. `ros2 bag record test`
2. failure `cannot import name 'import_c_library' from 'rpyutils'`

## System (please complete the following information)
 - OS: Ubuntu 20.04
 - ROS 2 Distro: Foxy
 - Version: Foxy stable

## Additional context
This used to work on foxy about a week ago.

ralph-lange, 2020-09-11:
I just got the same error after a fresh Foxy installation.

The problem: The Debian package ros-foxy-rpyutils is from 24 June 2020, but ros-foxy-rosbag2 is from 1 September 2020. Therefore, ros-foxy-rpytuils does not include the `import_c_library` from https://github.com/ros2/rpyutils/pull/4, which is used by rosbag2.

clalancette, 2020-09-11:
Thanks for digging in.  @jacobperron FYI, looks like we need to do a new release of rpyutils into Foxy and a new sync.

jacobperron, 2020-09-11:
Thanks for reporting! I've made a new release of rpyutils (https://github.com/ros/rosdistro/pull/26537) and will do another Foxy sync shortly.

jacobperron, 2020-09-11:
I've also added a new test (https://github.com/ros2/rosbag2/pull/523) in an effort to catch this kind of bug in earlier in the release cycle.

christianrauch, 2020-09-11:
Thanks for digging into this and providing a fix so promptly.

I am still surprised to see core tooling break during a stable release. This diminishes the trust of using ROS2 stable releases for serious projects.
Wouldn't it be possible to postpone such updates to the next stable release or at least try them out in testing and rolling releases before uploading them to the stable release?

crystaldust, 2020-09-14:
My PR also fails with the same logs(https://github.com/ros2/rclpy/pull/599), which seems to be the same problem. Hope the latest change will come to the foxy image ASAP

clalancette, 2020-09-14:
> Wouldn't it be possible to postpone such updates to the next stable release or at least try them out in testing and rolling releases before uploading them to the stable release?

That was all done.  The problem in this case is that we missed backporting and releasing an additional patch to rpyutils for Foxy.  It was further compounded by the fact that we didn't have any tests for this, which is now rectified by https://github.com/ros2/rosbag2/pull/523 .

In general, there is tension between keeping the released versions stable and adding fixes to the stable version.  To really be sure about patches that are backported, testing is needed.  That can come both in the form of automated tests, or manual testing by users.  Many automated tests are currently being added, so that side of the equation is at least being worked on.  The manual testing is (and has always been) a problem; there has historically been very little (I would say almost zero) testing of the "testing" repositories by the community.  I'm not sure how to improve that, but I'd definitely be open to suggestions.

jacobperron, 2020-09-14:
I believe this issue has been resolved. Feel free to comment or re-open.

closed 2020-09-14 by jacobperron

FilipDziwulski, 2020-09-16:
I'm seeing this failure when building in docker using:
`FROM ros:foxy`

jacobperron, 2020-09-16:
@FilipDziwulski I don't have an idea for how long it takes for the docker file to be updated, but you should be able to upgrade within the docker container to the newest package versions, e.g. `apt update && apt upgrade`.