ros2/rosbag2 #528: test_record__rmw_fastrtps_cpp failing

tfoote, 2020-10-02:
## Description
The nightly job has started failing for test_record using rmw_fastrtps_cpp

https://ci.ros2.org/job/nightly_linux_repeated/2063/testReport/(root)/projectroot/test_record__rmw_fastrtps_cpp/

### Expected Behavior
Nightly CI passes

### Actual Behavior
The CI failed with a segfault. 

```
[==========] Running 7 tests from 1 test case.
[----------] Global test environment set-up.
[----------] 7 tests from RecordIntegrationTestFixture
[ RUN      ] RecordIntegrationTestFixture.published_messages_from_multiple_topics_are_recorded
[NON-XML-CHAR-0x1B][37;1m2020-10-01 13:26:57.035 [NON-XML-CHAR-0x1B][31;1m[[NON-XML-CHAR-0x1B][37;1mPARTICIPANT[NON-XML-CHAR-0x1B][31;1m Error] [NON-XML-CHAR-0x1B][37mType with the same name already exists:rcl_interfaces::srv::dds_::GetParameters_Request_[NON-XML-CHAR-0x1B][34;1m -> Function [NON-XML-CHAR-0x1B][36mregisterType[NON-XML-CHAR-0x1B][m
[NON-XML-CHAR-0x1B][37;1m2020-10-01 13:26:57.035 [NON-XML-CHAR-0x1B][31;1m[[NON-XML-CHAR-0x1B][37;1mPARTICIPANT[NON-XML-CHAR-0x1B][31;1m Error] [NON-XML-CHAR-0x1B][37mType with the same name already exists:rcl_interfaces::srv::dds_::GetParameters_Response_[NON-XML-CHAR-0x1B][34;1m -> Function [NON-XML-CHAR-0x1B][36mregisterType[NON-XML-CHAR-0x1B][m
[INFO] [1601558817.041625662] [rosbag2_transport]: Listening for topics...
[INFO] [1601558817.045021523] [rosbag2_transport]: Subscribed to topic '/string_topic'
[INFO] [1601558817.146217994] [rosbag2_transport]: Subscribed to topic '/array_topic'
[INFO] [1601558817.146253032] [rosbag2_transport]: All requested topics are subscribed. Stopping discovery...
[       OK ] RecordIntegrationTestFixture.published_messages_from_multiple_topics_are_recorded (231 ms)
[ RUN      ] RecordIntegrationTestFixture.qos_is_stored_in_metadata
[INFO] [1601558817.264186644] [rosbag2_transport]: Listening for topics...
[INFO] [1601558817.365574545] [rosbag2_transport]: Subscribed to topic '/chatter'
[INFO] [1601558817.365610539] [rosbag2_transport]: All requested topics are subscribed. Stopping discovery...
[       OK ] RecordIntegrationTestFixture.qos_is_stored_in_metadata (215 ms)
[ RUN      ] RecordIntegrationTestFixture.records_sensor_data
[INFO] [1601558817.476706899] [rosbag2_transport]: Listening for topics...
[INFO] [1601558817.578261515] [rosbag2_transport]: Subscribed to topic '/string_topic'
[INFO] [1601558817.578293615] [rosbag2_transport]: All requested topics are subscribed. Stopping discovery...
[       OK ] RecordIntegrationTestFixture.records_sensor_data (112 ms)
[ RUN      ] RecordIntegrationTestFixture.receives_latched_messages
[INFO] [1601558817.590943212] [rosbag2_transport]: Listening for topics...
[INFO] [1601558817.592162399] [rosbag2_transport]: Subscribed to topic '/latched_topic'
[INFO] [1601558817.592512220] [rosbag2_transport]: All requested topics are subscribed. Stopping discovery...
-- run_test.py: return code -11
```

## To Reproduce
Run standard unit tests

## System (please complete the following information)
 - OS: Ubuntu
 - ROS 2 Nightly

nuclearsandwich, 2020-10-09:
This test stopped failing in https://ci.ros2.org/job/nightly_linux_repeated/2065/testReport/(root)/projectroot/test_record__rmw_fastrtps_cpp/ and has not failed since.

There was no change in rosbag2 between that build and the previous. Here is a diff of the vcs exact

```diff
--- nightly_linux_repeated-2064.repos   2020-10-09 12:37:16.378119233 -0700
+++ nightly_linux_repeated-2065.repos   2020-10-09 12:36:06.398116550 -0700
@@ -210,7 +210,7 @@
   ros2/launch_ros:
     type: git
     url: https://github.com/ros2/launch_ros.git
-    version: 8e34522a0087df36a6e2147a4c2a6788b6a58b09
+    version: 18be4d4fcf716e78739dc276f7e903b689cde4c2
   ros2/libyaml_vendor:
     type: git
     url: https://github.com/ros2/libyaml_vendor.git
@@ -246,7 +246,7 @@
   ros2/rcl_interfaces:
     type: git
     url: https://github.com/ros2/rcl_interfaces.git
-    version: 0dee9a4caf2865a7f9248ad000c1349205d0de5c
+    version: 9b7bf4b30e2f51c09fdac86ba711fd6a0f1a5910
   ros2/rcl_logging:
     type: git
     url: https://github.com/ros2/rcl_logging.git
@@ -266,7 +266,7 @@
   ros2/rcutils:
     type: git
     url: https://github.com/ros2/rcutils.git
-    version: 58580e86610f46867af8c4cabb56b86ad6ca2794
+    version: 945d30385bd76f3ea6c72f6164ddb64fba2a78f1
   ros2/realtime_support:
     type: git
     url: https://github.com/ros2/realtime_support.git
@@ -290,7 +290,7 @@
   ros2/rmw_fastrtps:
     type: git
     url: https://github.com/ros2/rmw_fastrtps.git
-    version: f887e5f22a8eb9a67d592fb69bc2679935386832
+    version: 27711b97b9ea8613df86f754be6639a49a52284c
   ros2/rmw_implementation:
     type: git
     url: https://github.com/ros2/rmw_implementation.git
@@ -346,7 +346,7 @@
   ros2/rosidl_typesupport_fastrtps:
     type: git
     url: https://github.com/ros2/rosidl_typesupport_fastrtps.git
-    version: 35b0021c365647c36d5c65c81e10103185e6caac
+    version: ca76c0e6701adfba7a2aa9b4185bf96ae36ded08
   ros2/rpyutils:
     type: git
     url: https://github.com/ros2/rpyutils.git
@@ -386,7 +386,7 @@
   ros2/unique_identifier_msgs:
     type: git
     url: https://github.com/ros2/unique_identifier_msgs.git
-    version: eb22cab2d963a8e548ab4adf6dd1bd913fcf53aa
+    version: 4ba68e5746490a1334fc6e2849e56be94cc5f27a
   ros2/urdf:
     type: git
     url: https://github.com/ros2/urdf.git
```

I am going to close this for now and it can be re-opened in the future if it recurs.

jacobperron, 2020-12-02:
This test is failing again: https://ci.ros2.org/job/nightly_linux_release/1752/testReport/(root)/rosbag2_transport/test_record__rmw_fastrtps_cpp_gtest_missing_result/

Not sure if it's a side-effect of the failure, but we see some cryptic error message at the end of the test:

```
04:11:28 30: [ RUN      ] RecordIntegrationTestFixture.receives_latched_messages
04:11:28 30: [INFO] [1606911088.541073276] [rosbag2_transport]: Listening for topics...
04:11:28 30: [INFO] [1606911088.542020279] [rosbag2_transport]: Subscribed to topic '/latched_topic'
04:11:28 30: [INFO] [1606911088.542224067] [rosbag2_transport]: All requested topics are subscribed. Stopping discovery...
04:11:28 30: unknown file: Failure
04:11:28 30: C++ exception with description "bad_function_call" thrown in the test body.
```

jacobperron, 2020-12-02:
I can reproduce the failure by stressing my machine and running the test repeatedly, e.g.

```
stress -c 32
```

```
while ./build/rosbag2_transport/test_record__rmw_fastrtps_cpp; do :; done
```

emersonknapp, 2020-12-02:
Thanks for the repro info, I am looking into it now.

emersonknapp, 2020-12-03:
Yeah - really easy to repro even without extra stress on my machine - i like to use the gtest args for that, pretty handy

```
./build/rosbag2_transport/test_record__rmw_fastrtps_cpp --gtest_repeat=-1 --gtest_break_on_failure
```

Rebuilding as debug, probably get into the meat of it tomorrow

closed 2020-12-04 by emersonknapp