ros2/rosbag2 #532: Is it possible to use rosbag2_py package as normal python package?

luqiang21, 2020-10-05:
I am trying to parse a ros2 bag and output some recorded info. Maybe I am not setting ros2 correctly, I tried to build rosbag2 repo using `colcon`, but always getting error: `ModuleNotFoundError: No module named 'ament_package' `. In my understanding, If I can build it correctly, I should be able to run rosbag2_py module, right? Besides this, is it possible to just install rosbag2_py package like other python package, so I can just import and use it?

Sorry if my question is too silly since I am new to ros2. Thank you for your help!

emersonknapp, 2020-10-09:
Do you have ROS2 installed (and sourced via setup.sh?) in your environment where you are trying to build rosbag2? Or, alternatively, do you have the whole ros2/ros2.repos file checked out in your workspace to build? One of these two should allow you to find ament_package.

luqiang21, 2020-10-10:
@emersonknapp I have ros2 dashing installed and sourced `/opt/ros/dashing/setup.bash` before`colcon build --merge-install`. But I am getting `Unknown CMake command "ament_export_targets".  ` immediately. Does this mean my ros2 is not installed properly? I created new workspace with rosbag2 in the `src` folder, then used colcon.

Karsten1987, 2020-10-11:
the python API is not available for Dashing. In fact, rosbag2_py is not released for any distro yet which means you might want to use [ROS2 from source](https://index.ros.org/doc/ros2/Installation/Latest-Development-Setup/). We might soon release it for the `rolling` distro, which provides binary installations.

luqiang21, 2020-10-12:
@Karsten1987 I see, thanks for clarification!

closed 2020-10-12 by luqiang21