ros2/rosbag2 #540: ros2 bag play fails with `Opened database for READ_ONLY`

fujitatomoya, 2020-10-16:
## Description

recording data via `ros2 bag record -a` cannot be played with `ros2 bag play <bag file directory>`

### Expected Behavior

`ros2 bag play <bag file directory>` plays messages.

### Actual Behavior

```
# ros2 bag play rosbag2_2020_10_16-12_39_09
[INFO] [1602819592.736418712] [rosbag2_storage]: Opened database 'rosbag2_2020_10_16-12_39_09/rosbag2_2020_10_16-12_39_09_0.db3' for READ_ONLY.
```

## To Reproduce

this only happens with source build but foxy. confirmed with https://github.com/ros2/ros2/commit/d2f6b73d7cce6dad5a76e4d4ac5036eb74aeb744.

```
> colcon build (or colcon build --symlink-install)
> source install/local_setup.bash
> ros2 topic pub /chatter std_msgs/String "data: Hello world" (terminal-1)
> ros2 bag record -a (terminal-2, wait 10sec and quit with Ctrl-C)
> (exit terminal-1)
> ros2 bag play <bag directory>(rosbag2_2020_10_16-12_39_09)
```

## System (please complete the following information)
 - OS: ubuntu20.04
 - ROS 2 Distro: Source build
 - Version: https://github.com/ros2/ros2/commit/d2f6b73d7cce6dad5a76e4d4ac5036eb74aeb744

## Additional context

Foxy does not have this issue.

fujitatomoya, 2020-10-16:
sqlite3 database does not have any data in `messages` table,

```
root@5c7a784cba7f:~/ros2_ws/colcon_ws# sqlite3 rosbag2_2020_10_16-12_39_09/rosbag2_2020_10_16-12_39_09_0.db3
SQLite version 3.31.1 2020-01-27 19:55:54
Enter ".help" for usage hints.
sqlite> .tables
messages  topics
sqlite> .dump topics
PRAGMA foreign_keys=OFF;
BEGIN TRANSACTION;
CREATE TABLE topics(id INTEGER PRIMARY KEY,name TEXT NOT NULL,type TEXT NOT NULL,serialization_format TEXT NOT NULL,offered_qos_profiles TEXT NOT NULL);
INSERT INTO topics VALUES(1,'/rosout','rcl_interfaces/msg/Log','cdr',replace('- history: 3\n  depth: 0\n  reliability: 1\n  durability: 1\n  deadline:\n    sec: 2147483647\n    nsec: 4294967295\n  lifespan:\n    sec: 10\n    nsec: 0\n  liveliness: 1\n  liveliness_lease_duration:\n    sec: 2147483647\n    nsec: 4294967295\n  avoid_ros_namespace_conventions: false\n- history: 3\n  depth: 0\n  reliability: 1\n  durability: 1\n  deadline:\n    sec: 2147483647\n    nsec: 4294967295\n  lifespan:\n    sec: 10\n    nsec: 0\n  liveliness: 1\n  liveliness_lease_duration:\n    sec: 2147483647\n    nsec: 4294967295\n  avoid_ros_namespace_conventions: false','\n',char(10)));
INSERT INTO topics VALUES(2,'/parameter_events','rcl_interfaces/msg/ParameterEvent','cdr',replace('- history: 3\n  depth: 0\n  reliability: 1\n  durability: 2\n  deadline:\n    sec: 2147483647\n    nsec: 4294967295\n  lifespan:\n    sec: 2147483647\n    nsec: 4294967295\n  liveliness: 1\n  liveliness_lease_duration:\n    sec: 2147483647\n    nsec: 4294967295\n  avoid_ros_namespace_conventions: false\n- history: 3\n  depth: 0\n  reliability: 1\n  durability: 2\n  deadline:\n    sec: 2147483647\n    nsec: 4294967295\n  lifespan:\n    sec: 2147483647\n    nsec: 4294967295\n  liveliness: 1\n  liveliness_lease_duration:\n    sec: 2147483647\n    nsec: 4294967295\n  avoid_ros_namespace_conventions: false','\n',char(10)));
INSERT INTO topics VALUES(3,'/chatter','std_msgs/msg/String','cdr',replace('- history: 3\n  depth: 0\n  reliability: 1\n  durability: 1\n  deadline:\n    sec: 2147483647\n    nsec: 4294967295\n  lifespan:\n    sec: 2147483647\n    nsec: 4294967295\n  liveliness: 1\n  liveliness_lease_duration:\n    sec: 2147483647\n    nsec: 4294967295\n  avoid_ros_namespace_conventions: false','\n',char(10)));
COMMIT;
sqlite> .dump messages
PRAGMA foreign_keys=OFF;
BEGIN TRANSACTION;
CREATE TABLE messages(id INTEGER PRIMARY KEY,topic_id INTEGER NOT NULL,timestamp INTEGER NOT NULL, data BLOB NOT NULL);
CREATE INDEX timestamp_idx ON messages (timestamp ASC);
COMMIT;
sqlite> select * from messages;
sqlite>
```

fujitatomoya, 2020-10-16:
CC: @Barry-Xu-2018

Barry-Xu-2018, 2020-10-16:
@fujitatomoya 

It is not a bug.   
There is a default cache before writing to disk.     
The default size is 1MB.   
The size of test message is too small so that nothing is writing to disk for long time.
You can change it to small size or 0.   
Such as  
ros2 bag record --max-cache-size 0 -o /tmp/111 /topic

Barry-Xu-2018, 2020-10-16:
There is a bug.  
While doing `ros2 bag record` and user want to exit, cached message should be written to disk.  
For small message and default cache size, current implementation leads that recorded messages are lost.   
So it is better to update bug description.

fujitatomoya, 2020-10-16:
I believe that with user aspect it should work in default setting out-of-box.

fujitatomoya, 2020-10-17:
@Barry-Xu-2018 

could you check https://github.com/ros2/rosbag2/pull/541?

fujitatomoya, 2020-10-17:
@Karsten1987 

friendly ping.

Barry-Xu-2018, 2020-10-17:
@fujitatomoya 

> could you check #541?

LGTM.

commit 1daf8334 (2020-10-20): if cache data exists, it needs to flush the data into the storage before shutdown (#541)

fix #540 
* if cache data exists, it needs to flush the data into the storage before shutdown.
* cache_ needs to reset after writing to storage.
* add reset_cache() method.
commit 1b3c71a3 (2020-11-02): if cache data exists, it needs to flush the data into the storage before shutdown (#541)

fix #540
* if cache data exists, it needs to flush the data into the storage before shutdown.
* cache_ needs to reset after writing to storage.
* add reset_cache() method.
commit 09ad5eac (2020-11-02): if cache data exists, it needs to flush the data into the storage before shutdown (#541)

fix #540
* if cache data exists, it needs to flush the data into the storage before shutdown.
* cache_ needs to reset after writing to storage.
* add reset_cache() method.
closed 2020-10-20 by jaisontj