ros2/rosbag2 #542: Windows build

d3nisdantas, 2020-10-19:
## Description
I'm trying to build rosbag2 in Windows. But getting the following build error when building rosbag_cpp:
`D:\tools\ros_ws\src\rosbag2\rosbag2_cpp\src\rosbag2_cpp\writers\sequential_writer.cpp(65): error C2039: 'MAX_BAGFILE_DURATION_NO_SPLIT': is not a member of 'rosbag2_storage::storage_interfaces' [D:\tools\ros_ws\src\build\rosbag2_cpp\rosbag2_cpp.vcxproj]
  D:\tools\ros2_foxy\include\rosbag2_storage/storage_factory.hpp(36): note: see declaration of 'rosbag2_storage::storage_interfaces'
D:\tools\ros_ws\src\rosbag2\rosbag2_cpp\src\rosbag2_cpp\writers\sequential_writer.cpp(65): error C2065: 'MAX_BAGFILE_DURATION_NO_SPLIT': undeclared identifier [D:\tools\ros_ws\src\build\rosbag2_cpp\rosbag2_cpp.vcxproj]
D:\tools\ros_ws\src\rosbag2\rosbag2_cpp\src\rosbag2_cpp\writers\sequential_writer.cpp(282): error C2039: 'MAX_BAGFILE_DURATION_NO_SPLIT': is not a member of 'rosbag2_storage::storage_interfaces' [D:\tools\ros_ws\src\build\rosbag2_cpp\rosbag2_cpp.vcxproj]
  D:\tools\ros2_foxy\include\rosbag2_storage/storage_factory.hpp(36): note: see declaration of 'rosbag2_storage::storage_interfaces'
D:\tools\ros_ws\src\rosbag2\rosbag2_cpp\src\rosbag2_cpp\writers\sequential_writer.cpp(282): error C2065: 'MAX_BAGFILE_DURATION_NO_SPLIT': undeclared identifier [D:\tools\ros_ws\src\build\rosbag2_cpp\rosbag2_cpp.vcxproj]
  Building Custom Rule D:/tools/ros2_foxy/src/gmock_vendor/CMakeLists.txt
  gtest-all.cc
  gmock-all.cc
  gmock.vcxproj -> D:\tools\ros_ws\src\build\rosbag2_cpp\gmock\Release\gmock.lib
  Building Custom Rule D:/tools/ros2_foxy/src/gmock_vendor/CMakeLists.txt
  gmock_main.cc
  gmock_main.vcxproj -> D:\tools\ros_ws\src\build\rosbag2_cpp\gmock\Release\gmock_main.lib
  Building Custom Rule D:/tools/ros_ws/src/rosbag2/rosbag2_cpp/CMakeLists.txt
  test_ros2_message.cpp
  typesupport_helpers.cpp
  introspection_message.cpp
     Creating library D:/tools/ros_ws/src/build/rosbag2_cpp/Release/test_ros2_message.lib and object D:/tools/ros_ws/src/build/rosbag2_cpp/Release/test_ros2_message.exp
  test_ros2_message.vcxproj -> D:\tools\ros_ws\src\build\rosbag2_cpp\Release\test_ros2_message.exe
Failed   <<< rosbag2_cpp [57.2s, exited with code 1]`

## To Reproduce
- Initialized Environment: "c:\Program Files (x86)\Microsoft Visual Studio\2017\Professional\VC\Auxiliary\Build\vcvars32.bat" 
- Sourced ROS2: "D:\tools\ros2_foxy\local_setup.bat"
- Build: colcon build --merge-install --event-handlers console_direct+

## System (please complete the following information)
- Windows 10
- Visual Studio 2017
- ROS2 foxy
- rosbag2 master
- cmake version 3.17.4

## Additional context
My goal is to be able to use the rosbag_py scripts to read ros2 bag in a windows environment. Would it be possible to do this in a more contained way, i.e., not having to depend on ROS2, or build all the rosbag packages?

emersonknapp, 2020-10-23:
We're not sure that rosbag2_py is compatible with the Foxy core, so you may not be able to build in this way - the package is newer than the Foxy release.

rosbag2_py is a thin layer over the rosbag2 C++ libraries - right now it is not modularized in such a way that you could only build the portion that understands how to read bags, unfortunately. Also, the messages are stored in a serialized format that rosbag2 doesn't have code to deserialize, in practice that happens by publishing the messages into the RMW system, where the subscribers know how to parse.

mjeronimo, 2020-11-02:
FYI, I've been working on porting rqt_bag to ROS2 and have found that rclpy has a serialization module that can serialize and deserialize ROS messages to the packed format used in the ROS2 rosbag. See https://github.com/ros-visualization/rqt_bag/blob/ros2/rqt_bag/src/rqt_bag/rosbag2.py#L107-L111 for an example. I have built and run the 'ros2' branch of rqt_bag on Windows, by the way, using the latest ROS2 code.

emersonknapp, 2020-11-02:
Yes, that's true! The rclpy layer can deserialize the messages - this combined with rosbag2_py to read the messages could get you what you need. However, this still depends on the ROS2 core.

@d3nisdantas i'm going to close this issue for now as not being a supported use case - please reopen if you think this is not correct

closed 2020-11-02 by emersonknapp