ros2/rosbag2 #555: Trouble building rosbag2 from source

evclark, 2020-10-31:
Creating a new issue for this as advised by @mabelzhang following our discussion in #308 

I have tried building from source as suggested, but am getting an error on the build. 

System: ROS2 Foxy installed by apt on Ubuntu 20.04 server on Raspberry Pi 4B (2GB version)

My process, following the instructions on the rosbag2 main README:
```
#Starting from within my project (named big_tree) ros2 workspace ~/big_tree/code/ros2_ws/
mkdir -p rosbag_ws/src
cd rosbag_ws/src
git clone https://github.com/ros2/rosbag2.git foxy
colcon build
```

Error output:
```
littlebird@rpi4b-0:~/big_tree/code/ros2_ws/src/rosbag_ws/src$ colcon build
Starting >>> sqlite3_vendor
[1.271s] WARNING:colcon.colcon_core.prefix_path.colcon:The path '/home/littlebird/big_tree/code/ros2_ws/src/install' in the environment variable COLCON_PREFIX_PATH doesn't exist
[1.271s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/littlebird/big_tree/code/ros2_ws/src/install/big_tree_python' in the environment variable AMENT_PREFIX_PATH doesn't exist
[1.272s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/littlebird/big_tree/code/ros2_ws/src/install/big_tree_msgs' in the environment variable AMENT_PREFIX_PATH doesn't exist
[1.272s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/littlebird/big_tree/code/ros2_ws/src/install/big_tree_msgs' in the environment variable CMAKE_PREFIX_PATH doesn't exist
Starting >>> shared_queues_vendor
Starting >>> rosbag2_test_common
Starting >>> zstd_vendor                                                                                             
Finished <<< sqlite3_vendor [7.41s]                                                                                                                                         
Finished <<< shared_queues_vendor [11.8s]                                                                                                        
Finished <<< rosbag2_test_common [12.2s]                                                                 
Starting >>> rosbag2_storage
[Processing: rosbag2_storage, zstd_vendor]                                                             
[Processing: rosbag2_storage, zstd_vendor]                                                                           
[Processing: rosbag2_storage, zstd_vendor]                                                                             
Finished <<< rosbag2_storage [1min 31s]                                                                                
Starting >>> rosbag2_storage_default_plugins
Finished <<< zstd_vendor [2min 7s]                                                                                               
Finished <<< rosbag2_storage_default_plugins [46.2s]                             
Starting >>> rosbag2_cpp
--- stderr: rosbag2_cpp                                
CMake Error at CMakeLists.txt:109 (find_package):
  By not providing "Findtest_msgs.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "test_msgs", but CMake did not find one.

  Could not find a package configuration file provided by "test_msgs" with
  any of the following names:

    test_msgsConfig.cmake
    test_msgs-config.cmake

  Add the installation prefix of "test_msgs" to CMAKE_PREFIX_PATH or set
  "test_msgs_DIR" to a directory containing one of the above files.  If
  "test_msgs" provides a separate development package or SDK, be sure it has
  been installed.

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/colcon_core/executor/__init__.py", line 91, in __call__
    rc = await self.task(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/colcon_core/task/__init__.py", line 93, in __call__
    return await task_method(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/colcon_ros/task/ament_cmake/build.py", line 66, in build
    if await has_target(args.build_base, 'install'):
  File "/usr/lib/python3/dist-packages/colcon_cmake/task/cmake/__init__.py", line 57, in has_target
    return target in await get_makefile_targets(path)
  File "/usr/lib/python3/dist-packages/colcon_cmake/task/cmake/__init__.py", line 77, in get_makefile_targets
    output = await check_output([
  File "/usr/lib/python3/dist-packages/colcon_core/subprocess.py", line 109, in check_output
    assert not rc, \
AssertionError: Expected ['/usr/bin/cmake', '--build', '/home/littlebird/big_tree/code/ros2_ws/src/rosbag_ws/src/build/rosbag2_cpp', '--target', 'help'] to pass: make: *** No rule to make target 'help'.  Stop.

---
Failed   <<< rosbag2_cpp [4.79s, exited with code 1]

Summary: 6 packages finished [2min 36s]
  1 package failed: rosbag2_cpp
  1 package had stderr output: rosbag2_cpp
  8 packages not processed
Expected ['/usr/bin/cmake', '--build', '/home/littlebird/big_tree/code/ros2_ws/src/rosbag_ws/src/build/rosbag2_cpp', '--target', 'help'] to pass: make: *** No rule to make target 'help'.  Stop.

```

Any advice helping guide me in what I'm doing wrong here is most appreciated!

iuhilnehc-ynos, 2020-11-02:
There are two methods to fix the package missing.
- install the package manually, such as `sudo apt-get install ros-foxy-test-msgs`
- use `rosdep install`, please refer to https://index.ros.org/doc/ros2/Installation/Foxy/Linux-Development-Setup/#install-dependencies-using-rosdep

BTW: git clone https://github.com/ros2/rosbag2.git `-b` foxy

jacobperron, 2020-11-02:
@evclark For this kind troubleshooting, I encourage you to ask your questions on [ROS Answers](https://answers.ros.org/). Check out our [guidelines for getting support](https://index.ros.org/doc/ros2/Contact/#help).

I'm going to close this since I don't think it is an issue with the code itself.

closed 2020-11-02 by jacobperron