ros2/rosbag2 #557: Recording with compression creates non-standard relative filepaths in metadata

jhdcs, 2020-11-04:
## Description
Running a bag record operation with compression enabled generates relative file paths that include the directory within the metadata. If compression is disabled, the relative file paths do not include the bag directory.

Ex: If recording a bag `bar` in the `foo` directory:

Filesystem View:
```
/../../foo/
       | bar/
       |  |  bar_0.sql.zstd
       |  |  bar_1.sql.zstd
       |  |  bar_2.sql.zstd
```

Relative file paths in metadata for compressed case:
- bar/bar_0.sql.zstd
- bar/bar_1.sql.zstd
- bar/bar_2.sql.zstd

Relative file paths in metadata for uncompressed case:
- bar_0.sql
- bar_1.sql
- bar_2.sql

### Expected Behavior
Consistent relative file path format within `metadata.yaml` across compression modes

### Actual Behavior
Relative filepath format within `metadata.yaml` differs between compression modes

## To Reproduce
1. Record uncompressed bag
2. Record compressed bag (use file compression, haven't checked this with message-based compression)
3. Compare `metadata.yaml` files between bags. The key of interest is `relative_file_paths`

## System (please complete the following information)
 - OS: Ubuntu Bionic
 - ROS 2 Distro: Rolling
 - Version: master branch (issue discovered on personal fork, however the compression and bag-writing code has not been modified, so this issue should be visible on master)

emersonknapp, 2020-11-20:
It seems like it plays back OK - it may be both writing and reading from an incorrect location based on using the same logic. Just a note

emersonknapp, 2021-01-08:
Ok - I think this is a bigger problem than I originally realized, I think it's affecting the ability to play back compressed backs correctly. Example case:

```
$ ros2 bag record --compression-mode message -a --compression-format zstd -o compressedbag
# record for a while
$ ls compressedbag
compressedbag_0.db3   metadata.yaml   
$ ros2 bag play compressedbag
[ERROR] [1610141873.094175026] [rosbag2_storage]: Could not open 'compressedbag' with 'sqlite3'. Error: Failed to setup storage. Error: Could not read-only open database. SQLite error (10): disk I/O error
[ERROR] [1610141873.094205949] [rosbag2_storage]: Could not load/open plugin with storage id 'sqlite3'.
[ERROR] [1610141873.094226227] [rosbag2_transport]: Failed to play: No storage could be initialized for: "compressedbag".
$ ros2 bag play compressedbag/compressedbag_0.db3
[INFO] [1610141876.190621674] [rosbag2_storage]: Opened database 'compressedbag/compressedbag_0.db3' for READ_ONLY.
```

Even if I manually fix the relative file paths listed in the metadata, this problem persists. This is definitely a bug and makes this feature unusable. Moving to top priority

PR #613 (2021-01-27): Fix relative metadata paths in SequentialCompressionWriter
Fixes #557
Related to #612 

This makes the SequentialCompressionWriter write in the bag metadata with correct relative paths in the V4 format.

* Use common init_metadata logic to get the correct relative path there
* In the file compression, be much more careful about what the paths truly mean, given that the metadata requires a path relative to the bag, but the compressor just needs to find and compress a file on the filesystem
* Factor out some common test logic so the new test can be a bit more readable
closed 2021-01-27 by emersonknapp