ros2/rosbag2 #559: Support for regex in defining recorded topics

TimoKettunen, 2020-11-11:
## Description
rosbag2 command should have an option -e to allow defining recording topics that match regular expression.
ROS(1) has this feature, ROS2 should have parity with ROS,
## Related Issues
None

## Completion Criteria
giving command 
```
$ ros2 bag record -e "<regex>" 
```
records all topics that match the regex

## Implementation Notes / Suggestions

## Testing Notes / Suggestions

emersonknapp, 2020-11-20:
This sounds like a neat feature!

adamdbrw, 2020-12-02:
Agreed, at least a couple of times I needed this as well

adamdbrw, 2021-01-19:
@TimoKettunen, @emersonknapp  turns out this would be very useful for our benchmarking package, so I submitted a PR draft. Hopefully this feature will soon be available.

adamdbrw, 2021-01-26:
This is resolved with #604

closed 2021-01-26 by adamdbrw