ros2/rosbag2 #560: Support for excluding topics

TimoKettunen, 2020-11-11:
## Description
rosbag2 command should have an option -x to allow excluding recording topics that match regular expression.
ROS(1) has this feature, ROS2 should have parity with ROS, See http://wiki.ros.org/rosbag/Commandline.
## Related Issues
None

## Completion Criteria
giving command 
```
$ ros2 bag record -a -x "<regex>" 
```
records all topics except those that match the regex

## Implementation Notes / Suggestions

## Testing Notes / Suggestions

emersonknapp, 2020-11-20:
Yes, confirmed that this is a feature gap. PRs welcome, not sure when we'll have bandwidth to take this on at the moment. It shouldn't be too difficult, probably a good first issue for someone.

adamdbrw, 2021-01-26:
This is resolved with #604

closed 2021-01-26 by adamdbrw