ros2/rosbag2 #561: Plan to port bag migration from ROS 1

weihunko, 2020-11-13:
## Description

In `rosbag` there is this functionality called [rosbag/migration](http://wiki.ros.org/rosbag/migration), which help reusing the old bag files when the message definitions change. 

## Related Issues

- https://github.com/ros2/rosbag2/issues/307

## Completion Criteria

- Port `rosbag/migration` or have a similar functionality in ROS 2

## Implementation Notes / Suggestions

- Port [rosbag/migration](http://wiki.ros.org/rosbag/migration)
- DDS also has this feature called [extensible type](https://www.omg.org/spec/DDS-XTypes/About-DDS-XTypes/), which could also be a potential solution

## Testing Notes / Suggestions

- None

emersonknapp, 2020-11-20:
Is the desired feature to migrate a ROS 1 bag to ROS 2, or to migrate a bag from an older ROS 2 distribution to newer message definitions?

Carlos-Loera, 2020-11-20:
@emersonknapp I can weigh in here. The latter; the ability to migrate and use an older bagfile's data from ROS 2 distribution to a new message definition.

emersonknapp, 2020-11-24:
Thanks for clarifying. Yes, I can see that this is a valuable use case to use your old data on a new release. 

I don't have a great sense now for what it would take to implement this, but I assume it requires having both distros installed and available. Given how different typesupport is in ros2 vs ros1, a direct port probably is not possible, it may require a lot of new code.

Carlos-Loera, 2021-01-20:
> Thanks for clarifying. Yes, I can see that this is a valuable use case to use your old data on a new release.
>
> I don't have a great sense now for what it would take to implement this, but I assume [...]

Are you able to provide me with your general scene on the expected timeline for such a tool to be implemented in ROS 2?

emersonknapp, 2021-01-21:
> Are you able to provide me with your general scene on the expected timeline for such a tool to be implemented in ROS 2?

There aren't any resources allocated for this currently. If you have an interest in working on it, I would be happy to help evaluate designs and code!

mschuckmann, 2021-11-04:
So if there is no ROS2 bag migration support does that mean that there is no way to use old bag files after a .msg file has been changed (e.g. it's package name changed or fields added or removed)?

emersonknapp, 2021-11-15:
It seems to be that is what it means, yes. There are probably some edge cases depending how CDR parsing is done by the DDS implementation, it might be able to parse messages even if there are new fields, but not entirely sure how this plays out in practice.

ozzdemir, 2023-08-11:
> It seems to be that is what it means, yes. There are probably some edge cases depending how CDR parsing is done by the DDS implementation, it might be able to parse messages even if there are new fields, but not entirely sure how this plays out in practice.

I've tried this case recently, when I try to echo, no message is received (With no warning etc, which surprised me); but when --raw flag is given to echo, I could observe the incoming bytes.
```SHELL
 ros2 topic echo /topic_msg_updated # Nothing is received, and no warning message is printed both in the bag play terminal and topic echo terminal
 ros2 topic echo /topic_msg_updated --raw # Could observe the incoming bytes.

```

MichaelOrlov, 2023-08-22:
- Closing this task since requested functionality will be implemented as part of the bigger feature implementation in core of the ROS 2  described in the [REP-2011](https://github.com/ros-infrastructure/rep/pull/358) and [REP-2016](https://github.com/ros-infrastructure/rep/pull/381) implementation.

BTW. We already have functionality for storing message definitions inside recorded bag files, however, don't have the ability yet to dynamically reconstruct type support and publisher from those stored message definitions.

closed 2023-09-01 by MichaelOrlov