ros2/rosbag2 #565: cannot record single hidden topic

albore, 2020-11-20:
## Description
`ros2 bag` doesn't record a single topic when this topic is hidden.

### Expected Behavior
To record rosbag of the single hidden topic.

### Actual Behavior
I tried to record the topic `/tree/snapshot_streams/_snapshots_0`, which is hidden. See the printout below.

```
$ ros2 topic list 
/battery/state
/led_strip/command
/parameter_events
/rosout
```

```
$ ros2 topic list --include-hidden-topics
/battery/state
/led_strip/command
/parameter_events
/rosout
/tree/snapshot_streams/_snapshots_0
```
Doing `ros2 topic echo /tree/snapshot_streams/_snapshots_0` I got data from the topic, regularly.

But `ros2 bag record /tree/snapshot_streams/_snapshots_0 --include-hidden-topics` does not subscribe to anything:

```
$ ros2 bag record /tree/snapshot_streams/_snapshots_0 --include-hidden-topics
[INFO] [1605875537.625338377] [rosbag2_storage]: Opened database 'rosbag2_2020_11_20-13_32_17/rosbag2_2020_11_20-13_32_17_0.db3' for READ_WRITE.
[INFO] [1605875537.632586883] [rosbag2_transport]: Listening for topics...
[WARN] [1605875537.632694785] [rosbag2_transport]: Hidden topics are not recorded. Enable them with --include-hidden-topics
```

## To Reproduce
** Steps to reproduce the behavior, e.g.
1. Publish information on a hidden topic.
2. ros2 bag record <topic_name> --include-hidden-topics

However, when using the _-a_ option, _all topics_ are recorded, including the hidden ones.

## System (please complete the following information)
 - OS: [e.g. Ubuntu Bionic]
 - ROS 2 Distro: Foxy
 - Version: ros-foxy-rosbag2 0.3.5-1focal.2020.11.03

emersonknapp, 2020-11-20:
This is an interesting use case, I think it just hadn't come up before, thanks for bringing it to our attention! I think the logic to fix it shouldn't be difficult, PRs are always welcome, otherwise we'll try to get to it for G-turtle

emersonknapp, 2021-05-14:
From #610 

Description
Hidden topics are not recorded by ros2 bag record if the topic name is explicitly provided as argument and the --include-hidden-topics is also enabled.

Actual behaviour
If I execute te next command to record a hidden topic (explicitly provided as argument)
ros2 bag record --output test_bag --include-hidden-topics /test_topic/_test_ &

I have the following response (specified topic is not listened):

[INFO] [1611325625.451160165] [rosbag2_storage]: Opened database 'test_bag/test_bag_0.db3' for READ_WRITE.
[INFO] [1611325625.667435684] [rosbag2_transport]: Listening for topics...
Then if I publish something to the topic
ros2 topic pub -1 /test_topic/_test_ std_msgs/UInt8 'data: 128'

I have the following response and nothing is recorded in the bag file

publisher: beginning loop
[WARN] [1611325639.692439400] [rosbag2_transport]: Hidden topics are not recorded. Enable them with --include-hidden-topics
publishing #1: std_msgs.msg.UInt8(data=128)
and nothing is recorded in the bag file.

Expected Behavior
After publishing something to the topic by
ros2 topic pub -1 /test_topic/_test_ std_msgs/UInt8 'data: 128'
the following response is provided.

[recordbag-1] [INFO] [1611325929.626733623] [rosbag2_transport]: Subscribed to topic '/test_topic/_test_'
[recordbag-1] [INFO] [1611325929.626775405] [rosbag2_transport]: All requested topics are subscribed. Stopping discovery...
instead of the warning and data is recorded in the bag file

System (please complete the following information)
OS: Ubuntu 20.04
ROS 2 Distro: ROS Foxy
rosbag2 version 0.4.0 (based on the below code I suppose the problem exist in 0.5.0 too)

Possible resolution
I had a look on the code and maybe the following modifications could solve the problem:
In rosbag2_transport/src/rosbag2_transport/recorder.cpp pass the include_hidden_topics parameter to the get_topics_with_types call.

std::unordered_map<std::string, std::string>
Recorder::get_requested_or_available_topics(
  const std::vector<std::string> & requested_topics,
  bool include_hidden_topics)
{
  return requested_topics.empty() ?
         node_->get_all_topics_with_types(include_hidden_topics) :
         node_->get_topics_with_types(requested_topics, include_hidden_topics);
}
In rosbag2_transport/src/rosbag2_transport/rosbag2_node.cpp (and .hpp) add new parameter to the get_topics_with_types

std::unordered_map<std::string, std::string> Rosbag2Node::get_topics_with_types(
  const std::vector<std::string> & topic_names,
  bool include_hidden_topics)
and at the end of the function definition call the filter_topics_with_more_than_one_type with passing the include_hidden_topics
return filter_topics_with_more_than_one_type(filtered_topics_and_types, include_hidden_topics));

emersonknapp, 2021-05-14:
Note: when fixed, this should be backported to Foxy

emersonknapp, 2021-08-09:
Fixed by #835

PR #835 (2021-08-09): Fixed inability to record hidden topics in foxy
Fixes #565
 
Ensure that the `--include-hidden-topics` flag is considered when collecting topics for recording.

This problem is already fixed on latest, and cannot be backported as it was part of a larger API-breaking change.
closed 2021-08-09 by emersonknapp