ros2/rosbag2 #583: Can't build master branch against rolling release

pjreed, 2020-12-09:
## Description
The master branch of this repository fails to compile in a fresh `ros:rolling` Docker container.  See logs below.

### Expected Behavior
Since the CI builds for this seem to be passing, I would expect this to compile in Ubuntu 20.04 / ROS Rolling, possibly only with also needing to clone any of its dependencies that have had API-breaking changes.  Still, this is reliably failing for me, and I don't see anything obvious that I could be missing.

### Actual Behavior
I created a `ros:rolling` Docker container, created a new workspace, cloned rosbag2 into it, installed its dependencies, and tried to build it.  That failed due to needing a new function in rcutils.  I then also cloned rcutils and installed its dependencies, then tried to clean and rebuild everything, but now `rosbag2_storage_default_plugins` fails to compile, and the failure does not seem to be due to any dependencies outside of the rosbag2 repository.  See steps below.

## To Reproduce

Run these steps:
```
docker run -it --rm ros:rolling
apt update && rosdep update
cd
mkdir -p src/rosbag2/src
cd src/rosbag2/src
git clone https://github.com/ros2/rosbag2.git
git clone https://github.com/ros2/rcutils.git
cd ..
rosdep install src --from-paths -i
colcon build
```

The complete build log is:
```
root@de5066246482:~/src/rosbag2# colcon build
Starting >>> rcutils 
Starting >>> sqlite3_vendor
Starting >>> shared_queues_vendor
Starting >>> zstd_vendor
Finished <<< sqlite3_vendor [0.74s]                                                                                                                         
Finished <<< shared_queues_vendor [2.16s]                                                                                            
Finished <<< rcutils [10.8s]                                                                    
Starting >>> rosbag2_test_common
Finished <<< rosbag2_test_common [1.21s]                                                              
Starting >>> rosbag2_storage
Finished <<< zstd_vendor [13.8s]                                                                       
Finished <<< rosbag2_storage [5.72s]                       
Starting >>> rosbag2_storage_default_plugins
--- stderr: rosbag2_storage_default_plugins                               
In file included from /root/src/rosbag2/src/rosbag2/rosbag2_storage_default_plugins/test/rosbag2_storage_default_plugins/sqlite/storage_test_fixture.hpp:36,
                 from /root/src/rosbag2/src/rosbag2/rosbag2_storage_default_plugins/test/rosbag2_storage_default_plugins/sqlite/test_sqlite_storage.cpp:29:
/root/src/rosbag2/src/rosbag2/rosbag2_storage_default_plugins/include/rosbag2_storage_default_plugins/sqlite/sqlite_storage.hpp:52:28: error: ‘StorageOptions’ in namespace ‘rosbag2_storage’ does not name a type
   52 |     const rosbag2_storage::StorageOptions & storage_options,
      |                            ^~~~~~~~~~~~~~
In file included from /root/src/rosbag2/src/rosbag2/rosbag2_storage_default_plugins/test/rosbag2_storage_default_plugins/sqlite/storage_test_fixture.hpp:36,
                 from /root/src/rosbag2/src/rosbag2/rosbag2_storage_default_plugins/test/rosbag2_storage_default_plugins/sqlite/test_sqlite_wrapper.cpp:27:
/root/src/rosbag2/src/rosbag2/rosbag2_storage_default_plugins/include/rosbag2_storage_default_plugins/sqlite/sqlite_storage.hpp:52:28: error: ‘StorageOptions’ in namespace ‘rosbag2_storage’ does not name a type
   52 |     const rosbag2_storage::StorageOptions & storage_options,
      |                            ^~~~~~~~~~~~~~
/root/src/rosbag2/src/rosbag2/rosbag2_storage_default_plugins/include/rosbag2_storage_default_plugins/sqlite/sqlite_storage.hpp:51:8: error: ‘void rosbag2_storage_plugins::SqliteStorage::open(const int&, rosbag2_storage::storage_interfaces::IOFlag)’ marked ‘override’, but does not override
   51 |   void open(
      |        ^~~~
In file included from /root/src/rosbag2/src/rosbag2/rosbag2_storage_default_plugins/test/rosbag2_storage_default_plugins/sqlite/test_sqlite_storage.cpp:29:
/root/src/rosbag2/src/rosbag2/rosbag2_storage_default_plugins/test/rosbag2_storage_default_plugins/sqlite/storage_test_fixture.hpp:140:20: error: ‘StorageOptions’ in namespace ‘rosbag2_storage’ does not name a type
  140 |   rosbag2_storage::StorageOptions make_storage_options_with_config(
      |                    ^~~~~~~~~~~~~~
/root/src/rosbag2/src/rosbag2/rosbag2_storage_default_plugins/include/rosbag2_storage_default_plugins/sqlite/sqlite_storage.hpp:51:8: error: ‘void rosbag2_storage_plugins::SqliteStorage::open(const int&, rosbag2_storage::storage_interfaces::IOFlag)’ marked ‘override’, but does not override
   51 |   void open(
      |        ^~~~
In file included from /root/src/rosbag2/src/rosbag2/rosbag2_storage_default_plugins/test/rosbag2_storage_default_plugins/sqlite/test_sqlite_wrapper.cpp:27:
/root/src/rosbag2/src/rosbag2/rosbag2_storage_default_plugins/test/rosbag2_storage_default_plugins/sqlite/storage_test_fixture.hpp:140:20: error: ‘StorageOptions’ in namespace ‘rosbag2_storage’ does not name a type
  140 |   rosbag2_storage::StorageOptions make_storage_options_with_config(
      |                    ^~~~~~~~~~~~~~
/root/src/rosbag2/src/rosbag2/rosbag2_storage_default_plugins/test/rosbag2_storage_default_plugins/sqlite/storage_test_fixture.hpp: In member function ‘void StorageTestFixture::write_messages_to_sqlite(std::vector<std::tuple<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, long int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >)’:
/root/src/rosbag2/src/rosbag2/rosbag2_storage_default_plugins/test/rosbag2_storage_default_plugins/sqlite/storage_test_fixture.hpp:103:49: error: cannot convert ‘<brace-enclosed initializer list>’ to ‘const string&’ {aka ‘const std::__cxx11::basic_string<char>&’}
  103 |     writable_storage->open({db_file, plugin_id_});
      |                                                 ^
[... 235 lines cut ...]
make[2]: *** [CMakeFiles/test_sqlite_wrapper.dir/build.make:63: CMakeFiles/test_sqlite_wrapper.dir/test/rosbag2_storage_default_plugins/sqlite/test_sqlite_wrapper.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:156: CMakeFiles/test_sqlite_wrapper.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/test_sqlite_storage.dir/build.make:63: CMakeFiles/test_sqlite_storage.dir/test/rosbag2_storage_default_plugins/sqlite/test_sqlite_storage.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:127: CMakeFiles/test_sqlite_storage.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< rosbag2_storage_default_plugins [5.14s, exited with code 2]

Summary: 6 packages finished [23.0s]
  1 package failed: rosbag2_storage_default_plugins
  1 package had stderr output: rosbag2_storage_default_plugins
  9 packages not processed
```

## System (please complete the following information)
 - OS: Ubuntu 20.04
 - ROS 2 Distro: ROS Rolling
 - Version: master

I also experience this problem in my host OS, but I replaced it inside a `ros:rolling` Docker container to be sure.

## Additional information

It seems as though perhaps this issue is happening because when colcon tries to build `rosbag2_storage_default_plugins`, it is including headers from the binary-installed `rosbag2_storage` package rather than the local workspace.  I'm not sure what needs to be done to make it include headers from the local workspace instead.

emersonknapp, 2020-12-09:
The reason for this is definitely that the build is including headers from the binary-installed `rosbag2` instead of from your source workspace, as seen in `In file included from /opt/ros/rolling/include/rosbag2_storage/storage_interfaces/read_write_interface.hpp:21`

I'm able to reproduce this easily enough. However, if you `apt-get remove ros-rolling-rosbag2-storage`, then your build is likely to succeed (it did for me).

Not sure exactly why this is happening, colcon should be preferring the version from the workspace. I know @Karsten1987 has also run into this before.

emersonknapp, 2020-12-09:
May be related to https://github.com/ros2/rosbag2/issues/447

pjreed, 2020-12-09:
Yeah, I think it is related to that.  I've got a hunch that this is just a matter of getting all of the ament export/dependencies lined up right between the packages and the tests, I'll see if I can figure it out.

emersonknapp, 2020-12-09:
Thanks! Your investigation is appreciated

pjreed, 2020-12-10:
I've got this *mostly* fixed, but `rosbag2_py` is turning out to be a bit tricky.

The catch is that it's using `pybind11_add_module` to declare its targets, like so:
```cmake
pybind11_add_module(_writer SHARED
  src/rosbag2_py/_writer.cpp
)
ament_target_dependencies(_writer PUBLIC
  "rosbag2_compression"
  "rosbag2_cpp"
  "rosbag2_storage"
)
clean_windows_flags(_writer)
```

It seems like `pybind11_add_module` is adding the binary-installed ROS path to the list of include paths first, and the paths added by `ament_target_dependencies` then get added second, so GCC prefer the earlier paths and fails.

For example, here's what the `c++` command currently looks like:
```
/usr/bin/c++  -DPLUGINLIB__DISABLE_BOOST_FUNCTIONS -DRCUTILS_ENABLE_FAULT_INJECTION -DROS_PACKAGE_NAME=\"rosbag2_py\" -D_writer_EXPORTS -I/usr/include/python3.8 -isystem /opt/ros/rolling/include -isystem /home/preed/src/rosbag2/install/rosbag2_compression/include -isystem /home/preed/src/rosbag2/install/rosbag2_cpp/include -isystem /home/preed/src/rosbag2/install/rosbag2_storage/include -isystem /home/preed/src/rosbag2/install/rcutils/include -isystem /opt/ros/rolling/opt/yaml_cpp_vendor/include -isystem /home/preed/src/rosbag2/install/zstd_vendor/include  -O2 -g -DNDEBUG -fPIC -fvisibility=hidden   -Wall -Wextra -Wpedantic -Werror -flto -fno-fat-lto-objects -std=gnu++14 -o CMakeFiles/_writer.dir/src/rosbag2_py/_writer.cpp.o -c /home/preed/src/rosbag2/src/rosbag2/rosbag2_py/src/rosbag2_py/_writer.cpp
```

Here's what it looks like if `ament_target_dependencies` is completely removed:
```
/usr/bin/c++  -DROS_PACKAGE_NAME=\"rosbag2_py\" -D_writer_EXPORTS -I/opt/ros/rolling/include -I/usr/include/python3.8  -O2 -g -DNDEBUG -fPIC -fvisibility=hidden   -Wall -Wextra -Wpedantic -Werror -flto -fno-fat-lto-objects -std=gnu++14 -o CMakeFiles/_writer.dir/src/rosbag2_py/_writer.cpp.o -c /home/preed/src/rosbag2/src/rosbag2/rosbag2_py/src/rosbag2_py/_writer.cpp
```

But here's what it looks like if I use `add_library` to declare the target instead, although this fails to build due to all the missing Python stuff:
```
/usr/bin/c++  -DPLUGINLIB__DISABLE_BOOST_FUNCTIONS -DRCUTILS_ENABLE_FAULT_INJECTION -DROS_PACKAGE_NAME=\"rosbag2_py\" -D_writer_EXPORTS -isystem /home/preed/src/rosbag2/install/rosbag2_compression/include -isystem /home/preed/src/rosbag2/install/rosbag2_cpp/include -isystem /home/preed/src/rosbag2/install/rosbag2_storage/include -isystem /opt/ros/rolling/include -isystem /home/preed/src/rosbag2/install/rcutils/include -isystem /opt/ros/rolling/opt/yaml_cpp_vendor/include -isystem /home/preed/src/rosbag2/install/zstd_vendor/include  -O2 -g -DNDEBUG -fPIC   -Wall -Wextra -Wpedantic -Werror -std=gnu++14 -o CMakeFiles/_writer.dir/src/rosbag2_py/_writer.cpp.o -c /home/preed/src/rosbag2/src/rosbag2/rosbag2_py/src/rosbag2_py/_writer.cpp
```

So `pybind11_add_module` is definitely responsible for adding `-isystem /opt/ros/rolling/include` to the beginning of the include list, and I'm not sure what the right way to handle that is.

Ideas that I have so far, none of which I like very much:
- Write our own version of `pybind11_add_module` that takes an argument with ament dependencies and adds them in the right order; this may break in the future if pybind11 changes
- Write our own `ament_target_dependencies` that manually manipulates the include path order to put its targets first, which will probably break if `ament_target_dependencies` changes in the future
- Just use `add_library` to declare the target and manually add all of the appropriate Python libraries and flags, which will probably result in a lot of ugly duplicated code between the three targets in `rosbag2_py`

Any other suggestions?

pjreed, 2021-01-28:
Well, I probably shouldn't have let this sit for so long -- I just went back to reproduce it again, and since there's been a new Rolling release since this originally happened and the Docker image has been updated, *most* of the problem has disappeared.  My branch in #585 fixes it, and #623 fixes it, too; maybe both are a good idea?

PR #585 (2021-01-29): Fix build issues when rosbag2_storage is binary installed
Several packages were failing to build from source when rosbag2_storage
was installed from a binary package because the path to the binary
install was being added to CMake's include path before the workspace
path.  This tweaks the dependencies and include orders to ensure the
source workspace path is preferred.

Fixes #583

Distro A, OPSEC #4584
commit 9890ee25 (2021-01-29): Fix build issues when rosbag2_storage is binary installed (#585)

Several packages were failing to build from source when rosbag2_storage
was installed from a binary package because the path to the binary
install was being added to CMake's include path before the workspace
path.  This tweaks the dependencies and include orders to ensure the
source workspace path is preferred.

Fixes #583

Distro A, OPSEC #4584
closed 2021-01-29 by emersonknapp