ros2/rosbag2 #586: `void PyEval_InitThreads()` is deprecated in Python 3.9

goekce, 2020-12-10:
## Description
Build of the rolling distribution fails because `rosbag2_py` cannot be build. Problem seems to be that `void PyEval_InitThreads()` is deprecated in Python 3.9

## To Reproduce
** Steps to reproduce the behavior, e.g.
1. compile the rolling distribution
2. Logs print error:

```
...
--- stderr: rosbag2_py
In file included from /opt/ros/pybind11_vendor/include/pybind11/cast.h:16,
                 from /opt/ros/pybind11_vendor/include/pybind11/attr.h:13,
                 from /opt/ros/pybind11_vendor/include/pybind11/pybind11.h:44,
                 from /opt/ros/src/ros2/rosbag2/rosbag2_py/src/rosbag2_py/./pybind11.hpp:24,
                 from /opt/ros/src/ros2/rosbag2/rosbag2_py/src/rosbag2_py/_storage.cpp:23:
/opt/ros/pybind11_vendor/include/pybind11/detail/internals.h: In function ‘pybind11::detail::internals& pybind11::detail::get_internals()’:
/opt/ros/pybind11_vendor/include/pybind11/detail/internals.h:276:28: error: ‘void PyEval_InitThreads()’ is deprecated [-Werror=deprecated-declarations]
  276 |         PyEval_InitThreads();
      |                            ^
In file included from /usr/include/python3.9/Python.h:145,
                 from /opt/ros/pybind11_vendor/include/pybind11/detail/common.h:112,
                 from /opt/ros/pybind11_vendor/include/pybind11/pytypes.h:12,
                 from /opt/ros/pybind11_vendor/include/pybind11/cast.h:13,
                 from /opt/ros/pybind11_vendor/include/pybind11/attr.h:13,
                 from /opt/ros/pybind11_vendor/include/pybind11/pybind11.h:44,
                 from /opt/ros/src/ros2/rosbag2/rosbag2_py/src/rosbag2_py/./pybind11.hpp:24,
                 from /opt/ros/src/ros2/rosbag2/rosbag2_py/src/rosbag2_py/_storage.cpp:23:
/usr/include/python3.9/ceval.h:130:37: note: declared here
  130 | Py_DEPRECATED(3.9) PyAPI_FUNC(void) PyEval_InitThreads(void);
      |                                     ^~~~~~~~~~~~~~~~~~
In file included from /opt/ros/pybind11_vendor/include/pybind11/cast.h:16,
                 from /opt/ros/pybind11_vendor/include/pybind11/attr.h:13,
                 from /opt/ros/pybind11_vendor/include/pybind11/pybind11.h:44,
                 from /opt/ros/src/ros2/rosbag2/rosbag2_py/src/rosbag2_py/./pybind11.hpp:24,
                 from /opt/ros/src/ros2/rosbag2/rosbag2_py/src/rosbag2_py/_writer.cpp:27:
/opt/ros/pybind11_vendor/include/pybind11/detail/internals.h: In function ‘pybind11::detail::internals& pybind11::detail::get_internals()’:
/opt/ros/pybind11_vendor/include/pybind11/detail/internals.h:276:28: error: ‘void PyEval_InitThreads()’ is deprecated [-Werror=deprecated-declarations]
...
cc1plus: all warnings being treated as errors
```

## System (please complete the following information)
 - OS: archlinux
 - ROS 2 Distro: Rolling
 - Version: master

## Additional context
** Add any other context about the problem here **

goekce, 2020-12-11:
probably this issue will be resolved when https://github.com/ros2/pybind11_vendor/issues/6 is resolved.

emersonknapp, 2020-12-11:
Thanks for reporting - it does look like it's internal pybind11 code so we should not need to make any changes here on the rosbag2 side to get the fix, once pybind11_vendor is upgraded

goekce, 2020-12-11:
Here is the corresponding PR https://github.com/ros2/pybind11_vendor/pull/8

emersonknapp, 2020-12-11:
Ah - it looks like that PR is blocked on development bandwidth - I will try to fit this on my backlog, though is teems unlikely I'll have time for it until the new year. Until then, community help would be valued.

clalancette, 2024-04-15:
This was fixed long ago with an update to pybind11_vendor, so closing this out.

closed 2024-04-15 by clalancette