ros2/rosbag2 #588: Support common trace format as storage type

windelbouwman, 2020-12-18:
## Description

It would be pretty cool if we could log data into the common trace format ([CTF](https://diamon.org/ctf/)). This is a format used by the linux LTTng tooling, and the format can be viewed in [trace compass](https://www.eclipse.org/tracecompass/), which is a graphical trace viewer.

This would enable:
- analysis of traces by trace compass and babel trace
- live trace forwarding using the `lttng-relay` program

## Steps involved

- translate topic types into CTF data format description
- automatically transform ROS2 messages into CTF events with the proper data format

emersonknapp, 2020-12-18:
This seems like a great candidate for a rosbag2_storage plugin, which could be implemented as a separate package. We don't currently deserialize at recording time, which means that translating the topics into your bag is nontrivial. It might be better written as a bag conversion after the fact? Just a thought.

windelbouwman, 2020-12-18:
That's correct, this is a great candidate for a rosbag2_storage plugin. This certainly is non-trivial. Decoding live might be cool, because this enables live view of the trace in a CTF live view capable tool. Since LTTng also emits CTF, the two traces could be combined, yielding good application insight!