ros2/rosbag2 #591: play a topic with a known type could be failure

iuhilnehc-ynos, 2020-12-23:
## Description

Using `ros2 bag` to play a specific topic name, which message type library was already known in the current environment, could lead to failure as the bag record file contains another unknown message type.

```shell
# ros2 bag play rosbag2_2020_12_23-05_16_14 --topics /robot_description
[INFO] [1608700982.524803618] [rosbag2_storage]: Opened database 'rosbag2_2020_12_23-05_16_14/rosbag2_2020_12_23-05_16_14_0.db3' for READ_ONLY.
[ERROR] [1608700982.525124475] [rosbag2_transport]: Failed to play: package 'dwb_msgs' not found, searching: [/opt/ros/foxy]
```

NOTE: The message type of topic `/robot_description` is `std_msgs/msg/String`.

### Expected Behavior
Play successfully

### Actual Behavior
`Failed to play: package 'dwb_msgs' not found`

## To Reproduce
Sorry for not introducing some detailed reproduce steps.

## System
 - OS: Ubuntu 20.04
 - ROS 2 Distro: Foxy
 - Version: 0bb6ab816c74a20bf488d66b84fc6c75b86b6c20

## Additional context
https://github.com/ros2/rosbag2/blob/0bb6ab816c74a20bf488d66b84fc6c75b86b6c20/rosbag2_cpp/src/rosbag2_cpp/readers/sequential_reader.cpp#L240-L243

Considering the time consumption of loading libraries and user experience, I think it's not necessary to check all topic types for a record file here.

emersonknapp, 2020-12-28:
Thanks for bringing this up! I agree that when `--topics` is specified for playback, the reader should not load all topic types, which will allow you to play the types you do know.

Another related useful feature would be `--ignore-unknown-topic-types` (or similar name) which is tracked at https://github.com/ros2/rosbag2/issues/307

It looks like you've started looking at the code a bit, a PR implementing this improvement would be very welcome.

iuhilnehc-ynos, 2020-12-29:
@emersonknapp 

Thanks for your reply.
About `--ignore-unknown-topic-types`, I prefer using [warning message](https://github.com/ros2/rosbag2/pull/394#discussion_r451228388) to adding a new option.
```
# ros2 bag play rosbag2_2020_12_29-05_14_33 
[INFO] [1609235758.153772574] [rosbag2_storage]: Opened database 'rosbag2_2020_12_29-05_14_33/rosbag2_2020_12_29-05_14_33_0.db3' for READ_ONLY.
[WARN] [1609235758.494371415] [rosbag2_transport]: Ignoring a topic '/particle_cloud', reason: package 'nav2_msgs' not found, searching: [/home/sharedata/Linux/ROS2/docker_ros2_master/local_install].
[WARN] [1609235758.597798745] [rosbag2_transport]: Ignoring a topic '/evaluation', reason: package 'dwb_msgs' not found, searching: [/home/sharedata/Linux/ROS2/docker_ros2_master/local_install].
[WARN] [1609235758.831430343] [rosbag2_transport]: Ignoring a topic '/global_costmap/costmap_raw', reason: package 'nav2_msgs' not found, searching: [/home/sharedata/Linux/ROS2/docker_ros2_master/local_install].
[WARN] [1609235758.886380607] [rosbag2_transport]: Ignoring a topic '/local_costmap/voxel_grid', reason: package 'nav2_msgs' not found, searching: [/home/sharedata/Linux/ROS2/docker_ros2_master/local_install].
[WARN] [1609235758.919309773] [rosbag2_transport]: Ignoring a topic '/local_costmap/costmap_raw', reason: package 'nav2_msgs' not found, searching: [/home/sharedata/Linux/ROS2/docker_ros2_master/local_install].
^C[INFO] [1609235761.409370679] [rclcpp]: signal_handler(signal_value=2)

```

I'll make a PR for these two features. (using --topics with a known type can be played; if a type does not exist, print a warning message)

PR #592 (2020-12-29): Fixed playing if unknown message types exist

Fixes #591

1. play a specific known message type even if some unknown types exist.
2. add a warning message while a message type library not exist.
closed 2020-12-29 by emersonknapp