ros2/rosbag2 #599: Ability to split already existing ros2 bags by time

Kaju-Bubanja, 2021-01-11:
## Description
In rosbag there is the ability to split already existing bags by time. That functionality would be useful for ros2 as well.

## Related Issues

None

## Completion Criteria

- A bag can be split by time

## Implementation Notes / Suggestions

Check how it was done in rosbag and how it is currently done for bags which get split by time which are recorded live.

## Testing Notes / Suggestions

Have a bag of 10s, split it after 5s to get 2 bags of 5s each.

vinnnyr, 2021-02-21:
@Kaju-Bubanja Can you please provide the example of the syntax the end user does to perform this in ROS1? I checked this [wiki](http://wiki.ros.org/rosbag/Commandline) and could not identify this feature.

Kaju-Bubanja, 2021-02-21:
See: https://answers.ros.org/question/99711/how-to-split-a-recorded-rosbag-file/

emersonknapp, 2023-01-20:
This feature (as worded) is also available in rosbag2 without external packages - seehttps://github.com/ros2/rosbag2#converting-bags for use of the `ros2 bag convert` command, which can provide the `max_bagfile_duration` argument for output bags.

However in verifying that this was so, I noticed that `ros2 bag convert` can't yet let you crop the time of a bagfile, writing out between start_time->end_time. This  information is currently in `PlayOptions` struct (that you fill via `ros2 bag play` arguments), but that is not part of the `convert` process right now. Allowing for this would be a useful improvement to that feature!

pfavr2, 2023-09-01:
I would love to help implement convert-start-offset (similar to play offset). Currently trying to understand how the .py and .cpp work. Any help would be greatly appreciated?

closed 2023-09-28 by MichaelOrlov