ros2/rosbag2 #610: Hidden topics are not recorded by 'ros2 bag record' when hidden topic name is provided as command line argument

goldminenine, 2021-01-22:
### Description
Hidden topics are not recorded by ros2 bag record if the topic name is explicitly provided as argument and the --include-hidden-topics is also enabled. 

### Actual behaviour
If I execute te next command to record a hidden topic (explicitly provided as argument)  
`ros2 bag record --output test_bag --include-hidden-topics /test_topic/_test_ & `

I have the following response (specified topic is not listened): 
```
[INFO] [1611325625.451160165] [rosbag2_storage]: Opened database 'test_bag/test_bag_0.db3' for READ_WRITE.
[INFO] [1611325625.667435684] [rosbag2_transport]: Listening for topics...
```

Then if I publish something to the topic
`ros2 topic pub -1 /test_topic/_test_ std_msgs/UInt8 'data: 128'`

I have the following response and nothing is recorded in the bag file 
```
publisher: beginning loop
[WARN] [1611325639.692439400] [rosbag2_transport]: Hidden topics are not recorded. Enable them with --include-hidden-topics
publishing #1: std_msgs.msg.UInt8(data=128)
```
and nothing is recorded in the bag file.

### Expected Behavior
 After publishing something to the topic by 
`ros2 topic pub -1 /test_topic/_test_ std_msgs/UInt8 'data: 128'`
the following response is provided.
```
[recordbag-1] [INFO] [1611325929.626733623] [rosbag2_transport]: Subscribed to topic '/test_topic/_test_'
[recordbag-1] [INFO] [1611325929.626775405] [rosbag2_transport]: All requested topics are subscribed. Stopping discovery...
```
instead of the warning and data is recorded in the bag file

### System (please complete the following information)
OS: Ubuntu 20.04
ROS 2 Distro: ROS Foxy
rosbag2 version 0.4.0 (based on the below code I suppose the problem exist in 0.5.0 too)

### Possible resolution
I had a look on the code and maybe the following modifications could solve the problem:
In rosbag2_transport/src/rosbag2_transport/recorder.cpp pass the include_hidden_topics parameter to the get_topics_with_types call.
```
std::unordered_map<std::string, std::string>
Recorder::get_requested_or_available_topics(
  const std::vector<std::string> & requested_topics,
  bool include_hidden_topics)
{
  return requested_topics.empty() ?
         node_->get_all_topics_with_types(include_hidden_topics) :
         node_->get_topics_with_types(requested_topics, include_hidden_topics);
}
```

In rosbag2_transport/src/rosbag2_transport/rosbag2_node.cpp (and .hpp) add new parameter to the get_topics_with_types
```
std::unordered_map<std::string, std::string> Rosbag2Node::get_topics_with_types(
  const std::vector<std::string> & topic_names,
  bool include_hidden_topics)
```
and at the end of the function definition call the filter_topics_with_more_than_one_type with passing the include_hidden_topics
` return filter_topics_with_more_than_one_type(filtered_topics_and_types, include_hidden_topics));`

emersonknapp, 2021-05-14:
Note: when this is fixed - it should probably also be backported to Foxy

emersonknapp, 2021-05-14:
Closing as duplicate of #565

closed 2021-05-14 by emersonknapp