ros2/rosbag2 #611: 👩‍🌾 Windows nightly failed due to shared_queues_vendor download failure

chapulina, 2021-01-22:
## Description

ROS 2's latest Windows nightly failed to build because of the `shared_queues_vendor` package. The failure happened while downloading this external project:

https://github.com/ros2/rosbag2/blob/de1158949dd969d7b69a30c7538889c42693a389/shared_queues_vendor/CMakeLists.txt#L21-L31

<details>
  <summary>Failure log</summary>
  
```
07:45:57 Not searching for unused variables given on the command line.
07:45:57 -- Selecting Windows SDK version 10.0.18362.0 to target Windows 10.0.18363.
07:45:57 -- The C compiler identification is MSVC 19.28.29336.0
07:45:57 -- The CXX compiler identification is MSVC 19.28.29336.0
07:45:57 -- Detecting C compiler ABI info
07:45:57 -- Detecting C compiler ABI info - done
07:45:57 -- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/VC/Tools/MSVC/14.28.29333/bin/Hostx64/x64/cl.exe - skipped
07:45:57 -- Detecting C compile features
07:45:57 -- Detecting C compile features - done
07:45:57 -- Detecting CXX compiler ABI info
07:45:57 -- Detecting CXX compiler ABI info - done
07:45:57 -- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/VC/Tools/MSVC/14.28.29333/bin/Hostx64/x64/cl.exe - skipped
07:45:57 -- Detecting CXX compile features
07:45:57 -- Detecting CXX compile features - done
07:45:57 -- Found ament_cmake: 1.0.3 (C:/ci/ws/install/share/ament_cmake/cmake)
07:45:57 -- Found PythonInterp: C:/Python38/python.exe (found suitable version "3.8.3", minimum required is "3") 
07:45:57 -- Using PYTHON_EXECUTABLE: C:/Python38/python.exe
07:45:57 -- Configuring done
07:45:57 -- Generating done
07:45:57 -- Build files have been written to: C:/ci/ws/build/shared_queues_vendor
07:45:57 Microsoft (R) Build Engine version 16.8.3+39993bd9d for .NET Framework

07:45:57 Copyright (C) Microsoft Corporation. All rights reserved.

07:45:57 

07:45:57   Checking Build System

07:45:57   Creating directories for 'ext-concurrentqueue'

07:45:57   Performing download step (download, verify and extract) for 'ext-concurrentqueue'

07:45:57   -- Downloading...

07:45:57      dst='C:/ci/ws/build/shared_queues_vendor/download/8f65a8734d77c3cc00d74c0532efca872931d3ce.zip'

07:45:57      timeout='60 seconds'

07:45:57      inactivity timeout='none'

07:45:57   -- Using src='https://github.com/cameron314/concurrentqueue/archive/8f65a8734d77c3cc00d74c0532efca872931d3ce.zip'

07:45:57   -- [download 100% complete]

07:45:57   -- [download 0% complete]

07:45:57   -- [download 1% complete]

07:45:57   -- [download 2% complete]

07:45:57   -- [download 3% complete]

07:45:57   -- [download 4% complete]

07:45:57   -- [download 5% complete]

07:45:57   -- [download 6% complete]

07:45:57   -- [download 7% complete]

[... 1295 lines cut ...]

The failure log above shows that the download started and stopped halfway. It was then attempted a second time and that doesn't seem to have gone through either. I couldn't understand from the logs what made the download fail, maybe someone else can parse the logs better. This could be a spurious failure, so I think it's just important to keep a close eye on it.

### Expected Behavior

Build completes 100% of the time

### Actual Behavior

Build failed

## To Reproduce

These builds run every night, so we can keep an eye on this job to see if it breaks again for the same reason:

https://ci.ros2.org/view/nightly/job/nightly_win_rel/

## System (please complete the following information)
 - OS: Windows
 - ROS 2 Distro: rolling
 - Version: master

---

https://github.com/osrf/buildfarmer/issues/145

Karsten1987, 2021-01-22:
I am having a little bit of trouble understanding what rosbag2 can do here. Do other builds fail as well? The vendor package is used by all three other platforms.

chapulina, 2021-01-23:
> Do other builds fail as well?

Only Windows failed. It looks like it was a one-time thing.

> I am having a little bit of trouble understanding what rosbag2 can do here.

If this becomes a common occurrence, maybe reconsidering the use of `ExternalProject_Add` could be an option. No downloads, no download failure.

emersonknapp, 2021-01-29:
Closing for now, builds are passing. If it shows up again we can reopen.

closed 2021-01-29 by emersonknapp