ros2/rosbag2 #618: Start playing messages after timestamp

Kaju-Bubanja, 2021-01-27:
## Description

A [full fledged player](https://github.com/ros2/rosbag2/issues/484) would be the optimal solution, but already being able to only start a bag after a certain timestamp can be useful. I propose to add a command line option --start_time timestamp and then to skip all messages until after that timestamp.

## Related Issues

This feature is a subset of the idea for a [full fledged player](https://github.com/ros2/rosbag2/issues/484).
This feature would be really useful in tandem with the [pause feature](https://github.com/ros2/rosbag2/pull/443)

## Completion Criteria

Only messages after the passed timestamp are published.

## Implementation Notes / Suggestions

The implementation seems really simple, basically only one line, it might not be the most efficient, because the not used messages still get loaded, but it's the most simple change that came to my mind, namely adding the if statement to the Player::play_messages_until_queue_empty function:

```
while (message_queue_.try_dequeue(message) && rclcpp::ok()) {
      if(options.start_time < message.message->time_stamp) {
          std::this_thread::sleep_until(
                  start_time_ + std::chrono::duration_cast<std::chrono::nanoseconds>(
                          1.0 / rate * message.time_since_start));
          if (rclcpp::ok()) {
              auto publisher_iter = publishers_.find(message.message->topic_name);
              if (publisher_iter != publishers_.end()) {
                  publisher_iter->second->publish(message.message->serialized_data);
              }
          }
      }
  }
```

## Testing Notes / Suggestions

A simple bag with some message play it with a starting time and check that all published messages have timestamps after the specified timestamp.

I am kinda stuck on how to add the option to the play_options.hpp file. I added following to play.py in rosbag2/ros2bag/verb/play.py:

```
parser.add_argument(
            '--start_time', type=check_positive_float, default=1.0,
            help='start timestamp from which to play the bag')
```

## Questions

Now there seems to be some importing stuff going on in rosbag2_transport/rosbag2_transport/\_\_init\_\_.py, namely: 
`rosbag2_transport_py = import_c_library('._rosbag2_transport_py', package='rosbag2_transport')`

but I haven't been able to find the module ._rosbag2_transport_py, I guess this is some autogenerated file by configure_python.cmake? How would I go about adding the start_time option to the PlayOptions struct? Also where is the cpp file for the PlayOptions? Also some options in play.py are not available in play_options.hpp for example uri. If somebody could help me shed some light on the process of how to add a config option that would be amazing.

emersonknapp, 2021-01-29:
This probably ties in with #484 as well, I think we need a design for a more fully featured Reader API, which could be taken advantage of by GUI/CLI any other type of tool, not just playback

KeisukeShima, 2021-07-13:
@Kaju-Bubanja @emersonknapp 
Hi, I am interested in this feature. Are there any updates regarding this feature or Random Access Reader?

Kaju-Bubanja, 2021-07-13:
You might want to have a look at: https://github.com/ros2/rosbag2/issues/696 It's quite a list of features, a lot already done. It might contain what you are looking for. For the jump to a certain timepoint, this is not yet implemented, but looking at the progress so far, this is probably coming out soon.

closed 2021-07-13 by Kaju-Bubanja

MichaelOrlov, 2021-07-14:
@KeisukeShima @Kaju-Bubanja I see some growing interests in Player::jump/seek API. 
I've created #821 to track activities for this feature. It's currently in WIP stage and I am trying to priritize it.
I am a little bit slow down with progress since jump/seek API is more generic and it's a little bit tricky how to properly jump backward and update cache with sequential reader. 
cc: @emersonknapp

KeisukeShima, 2021-07-14:
@MichaelOrlov Thank you for your comment. I appreciate your work.
I think the jump/seek API will be an important API for rosbag2 to improve usability.