ros2/rosbag2 #626: rosbag fails with ImportError

DavidATapia, 2021-01-29:
## Description
ros2 bag fails to record bag files

### Expected Behavior
ros2 bag can record bag files

### Actual Behavior
I tried to record on ROS2 foxy via ros2 bag record <log-file-name> which fails with:

```Traceback (most recent call last):
  File "/opt/ros/foxy/bin/ros2", line 11, in <module>
    load_entry_point('ros2cli==0.9.8', 'console_scripts', 'ros2')()
  File "/opt/ros/foxy/lib/python3.8/site-packages/ros2cli/cli.py", line 67, in main
    rc = extension.main(parser=parser, args=args)
  File "/opt/ros/foxy/lib/python3.8/site-packages/ros2bag/command/bag.py", line 38, in main
    return extension.main(args=args)
  File "/opt/ros/foxy/lib/python3.8/site-packages/ros2bag/verb/record.py", line 141, in main
    from rosbag2_transport import rosbag2_transport_py
  File "/opt/ros/foxy/lib/python3.8/site-packages/rosbag2_transport/__init__.py", line 17, in <module>
    rosbag2_transport_py = import_c_library('._rosbag2_transport_py', package='rosbag2_transport')
  File "/opt/ros/foxy/lib/python3.8/site-packages/rpyutils/import_c_library.py", line 39, in import_c_library
    return importlib.import_module(name, package=package)
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: /opt/ros/foxy/lib/librosbag2_storage.so: undefined symbol: _ZNK12class_loader11ClassLoader15isLibraryLoadedEv
The C extension '/opt/ros/foxy/lib/python3.8/site-packages/rosbag2_transport/_rosbag2_transport_py.cpython-38-x86_64-linux-gnu.so' failed to be imported while being present on the system. Please refer to 'https://index.ros.org/doc/ros2/Troubleshooting/Installation-Troubleshooting/#import-failing-even-with-library-present-on-the-system' for possible solutions```

## To Reproduce
** Steps to reproduce the behavior, e.g.
1. `ros2 bag record test`

## System (please complete the following information)OS: Ubuntu 20.04

 - OS: Ubuntu 20.04
 - ROS 2 Distro: Foxy
 - Version: 0.9.2-1focal.20210106.200442

## Additional context
This was working correctly with 0.9.2-1focal.20201103.164045, but performing a fresh install of ROS2 comes with this unexpected problem.  Also, `ros2 topic list` just hangs -- and I'm thinking the two problems might be related.

DavidATapia, 2021-02-01:
Hello @Karsten1987 and @dirk-thomas .  Has there been an opportunity to identify the issue or otherwise re-package the correct runtime library into the Debian packages?  I believe this problem might be affecting anyone who is currently installing ROS2 since version `0.9.2-1focal.20210106.200442`.  I see that you guys might be the maintainers for `ros-foxy-ros2cli` and `ros-foxy-rclpy` -- one of which is importing a bad `librosbag2_storage.so`

Much appreciated for your help!

clalancette, 2021-02-01:
For what it is worth, I can record bags with Foxy just fine here.  I am using a slightly newer version of `ros-foxy-ros-base`: 0.9.2-1focal.20210126.004113 .  You can get those packages from the ros2-testing repository by doing:

```
echo "deb http://packages.ros.org/ros2-testing/ubuntu focal main" | sudo tee /etc/apt/sources.list.d/ros2-latest.list
sudo apt-get update
sudo apt-get upgrade
```

Does that fix the problem for you?

DavidATapia, 2021-02-03:
@clalancette Thank you for the pointer to the `ros2-testing` repository.  As it turns out, the root cause of this issue is related to something we were doing incorrectly on our end.  My bad.

Our baseline incorporates the use of the `ros1_bridge`.  If the `ros1_bridge` has been built, then it is only acceptable to source the `./install/setup.bash`.  If the `ros1_bridge` has been built, and both `/opt/ros/foxy/setup.bash` and `./install/setup.bash` have been sourced (in that order) -- then we see the issue described above.

However, if the `ros1_bridge` has _not_ been built, then it is safe to source both `/opt/ros/foxy/setup.bash` and `./install/setup.bash` (though maybe that's bad practice?).  I'll go ahead and close this ticket.

Thanks again!

closed 2021-02-03 by DavidATapia