ros2/rosbag2 #629: QoS settings used for ros2 bag play

mjeronimo, 2021-02-01:
Documenting the following issue to get feedback and understand if it is worth addressing or if the case is so infrequent that a change like this is not worth considering...

## Description

When recording, ros2 bag record saves QoS information for each publisher on a specific topic to the database (under 'offered_qos_profiles'). Each message received on that topic is also saved. However, there is not an association between the messages and the QoS settings used to originally publish the messages. Currently, when playing back from a previously-recorded database, ros2 bag play creates a publisher for each topic (even if there were originally multiple publishers for that topic). When creating the publisher, it needs to provide a QoS profile. If the topic had a single publisher, or if a topic had multiple publishers but they all used the same QoS profile, there is not a problem figuring out the QoS profile to use. However, if the original publishers used different QoS profiles, ros2 bag play needs to adapt the recorded QoS profiles ("offered_qos_profiles") into one it can use for the publisher. In this case, the rosbag2 code uses a default QoS setting:

```C++
Rosbag2QoS Rosbag2QoS::adapt_offer_to_recorded_offers(
  const std::string & topic_name, const std::vector<Rosbag2QoS> & profiles)
{
  if (profiles.empty()) {
    return Rosbag2QoS{};
  }
  if (all_profiles_effectively_same(profiles)) {
    auto result = profiles[0];
    return result.default_history();
  }
  ROSBAG2_TRANSPORT_LOG_WARN_STREAM(
    "Not all original publishers on topic " << topic_name << " offered the same QoS profiles. "
      "Rosbag2 cannot yet choose an adapted profile to offer for this mixed case. "
      "Falling back to the rosbag2_transport default publisher offer.");
  return Rosbag2QoS{};
}
```
The result is that "ros2 bag play" is not quite faithful, from a QoS perspective to the original publishing. 

## Implementation Notes / Suggestions

To implement this case, we would need to know the QoS settings used for each published message. We currently store the QoS profiles, but they are not associated with individual messages, but are only associated to a topic (which could possibly have multiple publishers with different QoS). It may be that this is a rare situation and publishers of the same topic should be using the same QoS settings.

emersonknapp, 2021-02-03:
This is definitely something we considered when building it (and clearly just backed off from making a final decision). The problem here is that because ROS 2 is built up as an anonymized pub/sub system - we don't know where an individual message is coming from, and therefore don't know how to associate a specific message with a specific QoS.

There are a few cases to consider:
1. By querying all the QoS Profiles, you could create multiple subscriptions, each with one of the unique qos profiles. Any message coming in on the suscription could then be tagged with the associated QoS profile
    * you will (most likely) receive duplicate messages, because some subscriptions will likely be compatible with more than one of the publishers, given that most qos profiles are in a "strictness order" for compatibility - the least strict subscription is likely to receive a superset of the messages from a more strict subscription

2. By querying the exact origin ID of each message, we could then associate that message with an output publisher profile
    * Right now we have no way of knowing, through the RMW API, what participant a given message is coming from

3. We could construct a QoS profile for the single replay publisher, that would be "compatible with all original publishers"
    * This requires offering the "strictest common denominator" of mixed QoS profiles. This means that all original subscriptions on the given topic would be able to receive messages from the playback. _BUT_, this means that some original subscriptions that had a more strict requirement originally, would match with "all publishers on the topic" (the single playback publisher) rather than only a subset like it would have in the original system

Clearly option 1 is unacceptable (probably fully duplicatilng some topics' messages). Option 3 may be a decent compromise but could have subtly confusing side effects for users. Option 2 is the only truly correct way, but it also requires the deepest digging, and I'm not actually familiar enough with the DDS spec to know if it's possible

jacobperron, 2021-02-03:
Thanks @mjeronimo for opening the issue and @emersonknapp for laying out the options. I agree that option 2 seems the most appropriate.

I suggest we leave this ticket open for visibility, in case someone really wants to tackle it. Possibly label it as "backlog", as it doesn't sound like something we're planning to address.

emersonknapp, 2021-02-04:
Not sure if comment edits notify watchers, so I am commenting to note that I edited the above with a lot more promising context.

mjeronimo, 2021-02-05:
@emersonknapp Thanks for the helpful info. That'll give someone a good start on this.

MichaelOrlov, 2024-05-05:
- This feature request will be possible after #1553 will be implemented