ros2/rosbag2 #630: ROS2 Foxy Build Failure on WIN10 MSVC2019_x64 when compiling zstd_vendor

scottnortman, 2021-02-03:
## Description
Followed the instructions as listed here:  https://index.ros.org/doc/ros2/Installation/Foxy/Windows-Development-Setup/

### Expected Behavior
After typing the command:

> colcon test --merge-install

I received the following error:

c:\dev\ros2_foxy>colcon build --merge-install
Starting >>> ament_package
Starting >>> ament_lint
Starting >>> gtest_vendor
Starting >>> fastcdr
Starting >>> ament_cppcheck
Starting >>> cyclonedds
Starting >>> osrf_pycommon
Starting >>> osrf_testing_tools_cpp
Finished <<< ament_lint [4.66s]
Starting >>> ament_flake8
Finished <<< ament_cppcheck [4.86s]
Finished <<< ament_package [4.94s]
Starting >>> google_benchmark_vendor
Starting >>> ament_cmake_core
Finished <<< osrf_pycommon [5.06s]
Finished <<< gtest_vendor [5.20s]
Starting >>> gmock_vendor
Starting >>> urdfdom_headers
Finished <<< ament_flake8 [3.72s]
Starting >>> ament_pep257
Finished <<< gmock_vendor [3.78s]
Starting >>> ament_mypy
Finished <<< ament_cmake_core [7.33s]
Starting >>> ament_cmake_libraries
Finished <<< ament_pep257 [4.75s]
Starting >>> ament_cmake_export_libraries
Finished <<< ament_mypy [4.92s]
Starting >>> ament_cmake_python
Finished <<< ament_cmake_libraries [5.00s]
Starting >>> ament_cmake_export_dependencies
Finished <<< ament_cmake_export_libraries [5.11s]
Starting >>> ament_cmake_include_directories
Finished <<< ament_cmake_python [5.22s]
Starting >>> ament_cmake_test
Finished <<< ament_cmake_export_dependencies [5.95s]
Starting >>> ament_cmake_export_targets
Finished <<< ament_cmake_include_directories [5.39s]
Starting >>> ament_cmake_export_definitions
Finished <<< urdfdom_headers [20.5s]
Starting >>> ament_cmake_export_include_directories
Finished <<< ament_cmake_test [6.92s]
Starting >>> ament_cmake_export_link_flags
Finished <<< ament_cmake_export_targets [5.17s]
Starting >>> ament_cmake_version
Finished <<< ament_cmake_export_definitions [5.12s]
Starting >>> ament_cmake_export_interfaces
Finished <<< ament_cmake_export_include_directories [7.67s]
Starting >>> ament_cmake_target_dependencies
Finished <<< ament_cmake_export_link_flags [8.12s]
Starting >>> ament_copyright
Finished <<< ament_cmake_version [14.2s]
Starting >>> ament_cmake_gtest
Finished <<< ament_cmake_export_interfaces [14.3s]
Starting >>> ament_cmake_pytest
Finished <<< ament_copyright [11.9s]
Starting >>> domain_coordinator
Finished <<< ament_cmake_target_dependencies [15.2s]
Starting >>> ament_cmake
Finished <<< fastcdr [48.8s]
Starting >>> ament_index_python
Finished <<< ament_cmake_gtest [6.88s]
Starting >>> ament_cmake_gmock
Finished <<< ament_cmake_pytest [6.84s]
Starting >>> ament_lint_cmake
Finished <<< domain_coordinator [6.45s]
Starting >>> ament_xmllint
Finished <<< ament_lint_cmake [10.1s]
Starting >>> ament_cmake_lint_cmake
Finished <<< ament_index_python [11.6s]
Starting >>> ament_cpplint
Finished <<< ament_xmllint [8.38s]
Starting >>> rpyutils
Finished <<< ament_cmake [14.8s]
Finished <<< ament_cmake_gmock [14.0s]
Starting >>> uncrustify_vendor
Starting >>> ament_lint_auto
Finished <<< ament_cmake_lint_cmake [5.47s]
Starting >>> ament_cmake_copyright
Finished <<< ament_cpplint [5.41s]
Finished <<< rpyutils [5.00s]
Starting >>> launch
Starting >>> tinyxml2_vendor
Finished <<< google_benchmark_vendor [1min 2s]
Starting >>> ament_cmake_google_benchmark
Finished <<< ament_lint_auto [7.48s]
Starting >>> yaml_cpp_vendor
Finished <<< ament_cmake_copyright [6.27s]
Starting >>> ament_cmake_cppcheck
Finished <<< launch [7.67s]
Starting >>> ament_cmake_cpplint
Finished <<< ament_cmake_google_benchmark [8.70s]
Starting >>> ament_cmake_xmllint
Finished <<< osrf_testing_tools_cpp [1min 16s]
Starting >>> ament_cmake_flake8
Finished <<< ament_cmake_cppcheck [8.23s]
Starting >>> ament_cmake_pep257
Finished <<< ament_cmake_cpplint [8.41s]
Starting >>> launch_testing
Finished <<< ament_cmake_xmllint [8.19s]
Starting >>> foonathan_memory_vendor
Finished <<< ament_cmake_flake8 [9.50s]
Starting >>> eigen3_cmake_module
Finished <<< ament_cmake_pep257 [7.55s]
Starting >>> tinyxml_vendor
Finished <<< launch_testing [7.02s]
Starting >>> libcurl_vendor
Finished <<< cyclonedds [1min 30s]
Starting >>> shared_queues_vendor
Finished <<< tinyxml2_vendor [27.3s]
Starting >>> ament_cmake_auto
Finished <<< ament_cmake_auto [9.86s]
Starting >>> rviz_assimp_vendor
Finished <<< foonathan_memory_vendor [25.7s]
Starting >>> fastrtps
Finished <<< eigen3_cmake_module [30.5s]
Starting >>> orocos_kdl
Finished <<< tinyxml_vendor [29.3s]
Starting >>> rviz_ogre_vendor
Finished <<< shared_queues_vendor [45.7s]
Starting >>> zstd_vendor
Finished <<< yaml_cpp_vendor [1min 25s]
Starting >>> test_interface_files
Finished <<< test_interface_files [5.89s]
Starting >>> launch_xml
Finished <<< uncrustify_vendor [2min 5s]
Starting >>> ament_uncrustify
Finished <<< launch_xml [29.9s]
Starting >>> sqlite3_vendor
Finished <<< ament_uncrustify [5.06s]
Starting >>> ament_cmake_uncrustify
Finished <<< orocos_kdl [1min 23s]
Starting >>> tracetools_trace
Finished <<< ament_cmake_uncrustify [8.41s]
Starting >>> ament_lint_common
Finished <<< tracetools_trace [5.45s]
Starting >>> ament_pycodestyle
Finished <<< ament_pycodestyle [3.52s]
Starting >>> launch_yaml
Failed   <<< zstd_vendor [1min 13s, exited with code 1]
Aborted  <<< launch_yaml [1.14s]
Aborted  <<< ament_lint_common [7.61s]
Aborted  <<< sqlite3_vendor [1min 8s]
Aborted  <<< rviz_assimp_vendor [6min 4s]
Aborted  <<< fastrtps [7min 37s]
Aborted  <<< libcurl_vendor [9min 4s]
Aborted  <<< rviz_ogre_vendor [10min 58s]

Summary: 65 packages finished [13min 1s]
  1 package failed: zstd_vendor
  7 packages aborted: ament_lint_common fastrtps launch_yaml libcurl_vendor rviz_assimp_vendor rviz_ogre_vendor sqlite3_vendor
  236 packages not processed

sloretz, 2021-02-03:
Linking: https://answers.ros.org/question/370897/ros2-build-error-on-win10/

Which has more info:

```
-- Selecting Windows SDK version 10.0.18362.0 to target Windows 10.0.19041.
-- The C compiler identification is MSVC 19.25.28614.0
-- The CXX compiler identification is MSVC 19.25.28614.0
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.25.28610/bin/Hostx64/x64/cl.exe
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.25.28610/bin/Hostx64/x64/cl.exe - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.25.28610/bin/Hostx64/x64/cl.exe
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.25.28610/bin/Hostx64/x64/cl.exe - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found ament_cmake: 0.9.8 (C:/dev/ros2_foxy/install/share/ament_cmake/cmake)
-- Found PythonInterp: C:/Python38/python.exe (found suitable version "3.8.3", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: C:/Python38/python.exe
-- Configuring done
-- Generating done
-- Build files have been written to: C:/dev/ros2_foxy/build/zstd_vendor
Microsoft (R) Build Engine version 16.5.0+d4cbfca49 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

  Checking Build System
  Creating directories for 'zstd-1.4.4'
  Building Custom Rule C:/dev/ros2_foxy/src/ros2/rosbag2/zstd_vendor/CMakeLists.txt
  Performing download step (download, verify and extract) for 'zstd-1.4.4'
  -- Downloading...
     dst='C:/dev/ros2_foxy/build/zstd_vendor/zstd-1.4.4-prefix/src/v1.4.4.zip'
     timeout='60 seconds'
  -- Using src='https://github.com/facebook/zstd/archive/v1.4.4.zip'
  -- [download 100% complete]
  -- verifying file...
         file='C:/dev/ros2_foxy/build/zstd_vendor/zstd-1.4.4-prefix/src/v1.4.4.zip'
  -- Downloading... done
  -- extracting...
       src='C:/dev/ros2_foxy/build/zstd_vendor/zstd-1.4.4-prefix/src/v1.4.4.zip'
       dst='C:/dev/ros2_foxy/build/zstd_vendor/zstd-1.4.4-prefix/src/zstd-1.4.4'
  -- extracting... [tar xfz]
  -- extracting... [analysis]
  -- extracting... [rename]
  -- extracting... [clean up]
  -- extracting... done
  No update step for 'zstd-1.4.4'
  Performing patch step for 'zstd-1.4.4'
  'patch' is not recognized as an internal or external command,
  operable program or batch file.
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(231,5): error MSB6006: "cmd.exe" exited with code 9009. [C:\dev\ros2_foxy\build\zstd_vendor\zstd-1.4.4.vcxproj]
```

sloretz, 2021-02-03:
Appears to be caused by #579 @emersonknapp @clalancette FYI

scottnortman, 2021-02-03:
Suggested use of an alternative patch.exe as described from the alternate Stack Overflow post

https://stackoverflow.com/questions/517257/how-do-i-apply-a-diff-patch-on-windows

worked; after adding the specified folder to allow for command line calling of patch.exe I was able to build

Thanks for the quick help.

closed 2021-02-03 by scottnortman

clalancette, 2021-02-03:
We should actually fix that to use `git apply` instead of `patch`, since `patch` needs Administrator on Windows.  I'll open a low-priority issue to fix it.